MarlinFirmware/Marlin #21589: [incomplete implementation] chamber fan constant fan used

GMagician, 2021-04-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm looking to activate some sort of fan to lower temp in chamber nad I found this in code:

https://github.com/MarlinFirmware/Marlin/blob/c35e271dba2f608cb72f48357a1d36a8bfa66a12/Marlin/src/module/temperature.cpp#L1411

I'm wondering is, in this case, it's better to recycle CHAMBER_FAN_INDEX or define this fan by changing `//#define CHAMBER_FAN              // Enable a fan on the chamber` in configuration.

I'll take care of post a PR but an hint is welcome

### Expected behavior

As code says...a configurable fan

### Actual behavior

fan 2 is always used

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Bear 2.1

### Electronics

AGCM4

### Add-ons

Chamber sensor and fan

### Your Slicer

Cura

### Host Software

Repetier Host

thisiskeithb, 2021-04-13:
Did you take care of everything in your PR? https://github.com/MarlinFirmware/Marlin/pull/21594

GMagician, 2021-04-13:
No, since I'm not sure if CHAMBER_FAN_INDEX may be recycled or if it's better to recycle CHAMBER_FAN.
Both are related to chamber fan but what former purpose?

GMagician, 2021-04-13:
It seems that there is some duplicated code involving CHAMBER_FAN_INDEX. This fans seems a sort of CHAMBER_FAN_MODE=0 but latter has no AUTO_POWER_CONTROL  support.

closed 2021-06-24 by github-actions[bot]