MarlinFirmware/Marlin #21591: Too long command line when compiling under Windows10

arka2pol, 2021-04-11:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Compiling under linux goes OK, but compiling under Windows gives a message:

(#define CONFIGURATION_H_VERSION 020007)

fork/exec C:\Users\andrzej.kowalski.RAC\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc.exe: The filename or extension is too long.
Błąd kompilacji  dla płytki (= compilation error for board) Arduino Mega or Mega 2560.

:-)

### Steps to Reproduce

just try to compile under Win10

### Version of Marlin Firmware

newest

### Printer model

Prusa compatible self made

### Electronics

rumba

### Add-ons

BL-Touch

### Your Slicer

Cura

### Host Software

SD Card (headless)

GMagician, 2021-04-11:
Not sure arduino is still supported as development tools but this "path too long" issue, I suspect, it needs you to move your source in a short path (eg c:\marlin)

bill-orange, 2021-04-11:
I agree with @GMagician .  I don't think the Arduino compiler is supported anymore.  I tried compiling for an 8-bit machine with the Arduino compiler a few months ago, as a test,  and had the same problem.

I think you are going to have to switch to VSCode.  Its just as well.  For a complex project like Marlin, VSCode is much more suitable, especially if you combine it with your own Github repository.  If I could not track and review my changes, I would be lost.

Liriel-666, 2021-04-11:
That's a normal arduino error and not marlin special. 
Because that you should use visual Code studio which is ways better

arka2pol, 2021-04-11:
Tried, (C:\A ), also tried one letter user name. No result

It seems it is no way to move temporary files somewhere else, always uses “….appdata/local…….”

 

(by the way – please propose another compiler?)

arka2pol, 2021-04-11:
Will try….. (wrrrrrr, I like Arduino)

closed 2021-04-11 by thisiskeithb