MarlinFirmware/Marlin #21593: XYZ Homes in wrong direction (max position) [BUG] (bug summary)

D1TG, 2021-04-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

XYZ motors all move in the correct direction when testing. Only when Homed they will move in the wrong direction, even after I deliberately inverted motor direction. 

### Expected behavior

home move to min endstops X,Y and Z

### Actual behavior

homes moves to max where there is no endstop and Sensorless homing stops working too

### Steps to Reproduce

press Home command 

### Version of Marlin Firmware

Merlin 2.0

### Printer model

JG Aurora A5

### Electronics

BTT SKR 1.4 Turbo BTT TFT35-E3 v3.0

### Host Software

Pronterface

ellensp, 2021-04-11:
Please attach your configuration files.

D1TG, 2021-04-12:
**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

#define CONFIG_EXAMPLES_DIR "JGAurora/A5"

/**
 * JGAurora A5 configuration
 * Authors: Telli Mantelli, Kris Waclawski, Michael Gilardi & Samuel Pinches
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 020007

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
[... 2326 lines cut ...]
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

ellensp, 2021-04-12:
just zip up configs and attach.  (drag and drop into reply)

D1TG, 2021-04-12:
im unable to upload the file,even in .txt ?

ellensp, 2021-04-12:
you did, but its long and annoying and not easy to use in testing.

D1TG, 2021-04-12:
I HOPE THIS WORKD. My configuration.h on .txt 

[configH.txt](https://github.com/MarlinFirmware/Marlin/files/6298506/configH.txt)

jameswis01, 2021-04-23:
Im having almost the Exact same issue.  I've had someone else duplicate it.  
If Disable Endstops, then X&Y move in correct direction, but sensorless homing stops working.
Endstops enabled: X&Y Homes in opposite direction of the X+/X- and Y+/Y- buttons.
Attaching my Config.h and Advconfig.

[Bugfix configs.zip](https://github.com/MarlinFirmware/Marlin/files/6361934/Bugfix.configs.zip)

closed 2021-07-04 by github-actions[bot]