MarlinFirmware/Marlin #21596: [BUG] reported temp miss decimals

GMagician, 2021-04-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Repetier reports temperatures without decimals (or better decimals are always 0).
I'm not sure if this is intentional but i found that bed temperature is returned as `celsius_t` that is an `int16_t` by `degBed` and `degTargetBed` but print functiont needs a float.

Extruders always reports the same (decimals 0) even if float are returned by `degHotend` and `degTargetHotend` (in this case is the returned variable that is `celsius_t`)

### Expected behavior

Temperature with decimals

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Bear 2.1

### Electronics

AGCM4

### Your Slicer

Cura

### Host Software

Repetier Host

GMagician, 2021-04-11:
searching I saw #21519 will address it. I keep it active as a note for other

Pyro-Fox, 2021-04-11:
+1 on this one. Thank you @GMagician for finding the PR. I was about to open this same bug.

thisiskeithb, 2021-04-12:
I added a `Bug: Confirmed!` label even though the change was intentional.

We’ll see what happens in the linked PR.

ramiropolla, 2021-04-12:
I think it all boils down to this change:
https://github.com/MarlinFirmware/Marlin/commit/e5ff55a1be7646b6159e6dedac50bfbe57e6dfa0#r49270930

thinkyhead, 2021-04-16:
Hosts should be able to handle values without `.0` on the end. When we have something other than `.0` to emit in that spot it will reappear. But all temperature handling in Marlin is done in coarse, integer, whole celsius degree units.

closed 2021-04-16 by thinkyhead

GMagician, 2021-04-16:
Hosts handle correctly integer temp values (marlin reports them always as decimal values followed by .00) The odd things is only graph you get