MarlinFirmware/Marlin #21598: [BUG] 'manual_feedrate_mm_s' was not declared in this scope; did you mean 'feedrate_mm_s'?

arbitur, 2021-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I cannot build because definitions seems to not be defined in the correct order. Here are [my config files](https://github.com/MarlinFirmware/Marlin/files/6293329/Configurations.zip)
The problem is caused by `manual_feedrate_mm_s` not being defined and it isnt defined because `MANUAL_FEEDRATE` isnt defined and it is not defined because neither `ULTIPANEL and EXTENSIBLE_UI` are defined and Im not supposed to define `ULTIPANEL` because I have `REPRAP_DISCOUNT_SMART_CONTROLLER`. Before commit f6ffbe548c81abfeecc996c8f2bffd27cabe0014 it worked because it would define `ULTIPANEL` by defining `IS_ULTIPANEL` when `REPRAP_DISCOUNT_SMART_CONTROLLER` was defined, but not after that commit.

### Bug Timeline

Since commit f6ffbe548c81abfeecc996c8f2bffd27cabe0014 

### Version of Marlin Firmware

latest bugfix-2.0.x

ellensp, 2021-04-12:
Config files provided are 020006, not for bugfix.

arbitur, 2021-04-12:
Oh oops, I forgot I was on the working commit when I zipped them. Ill update.

arbitur, 2021-04-12:
@ellensp Now Ive uploaded the correct files.

ellensp, 2021-04-12:
New provided Config files works as expected, no errors.

arbitur, 2021-04-12:
@ellensp Im terribly sorry Im very tired, the updated config files includes my proposed fix... Ill update one last time. Try now.

ellensp, 2021-04-12:
your configuration_adv.h line 1132 is incorrect
`#if EITHER(ULTIPANEL, EXTENSIBLE_UI)`
should be 
`#if EITHER(IS_ULTIPANEL, EXTENSIBLE_UI)`
as per the default config https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L1130

closed 2021-04-12 by ellensp

arbitur, 2021-04-12:
Wow it was during merge from bugfix remote to my local branch this problem emerged. Thanks.