MarlinFirmware/Marlin #21599: [BUG] (SKR E3 turbo sd detect pin is defined but auto detection is not working)

dmcleish91, 2021-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Updated my mainboard from an SKR E3 mini to an SKR E3 turbo to test out dual z configuration. noticed that sd detection is not working on new board. I see a defined sd detect pin for this LPC1769 board and #define SDCARD_CONNECTION ONBOARD is enabled so I'm not sure why I'm only getting the menu option to attach an sd card.

I have confirmed that this feature is working on the bugfix branch of marlin for the SKR E3 mini.

### Bug Timeline

new bug discovered 4/9/2021

### Expected behavior

insert properly formatted sd card and see media inserted message and have the ability to access the files. Open ejection the media removed is displayed 

### Actual behavior

inserted sd card and no message is displayed only option is to use menu item to attach sd card.

### Steps to Reproduce

1.) insert sd card
2.) observe no media inserted message

### Version of Marlin Firmware

bugfix 2.0.X

### Printer model

Creality Ender 3

### Electronics

SKR E3 Turbo

### Add-ons

second z stepper motor connected to e1 stepper driver

### Your Slicer

Prusa Slicer

ellensp, 2021-04-12:
Please attach your configuration files. Just in case you have something in the config that triggers the issue.

thisiskeithb, 2021-04-12:
> Please attach your configuration files. Just in case you have something in the config that triggers the issue.

@ellensp: I’ve confirmed this on my printer here and with @dmcleish91 on Discord.

You can reproduce it with the E3 Turbo configs from the Configs repo: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo

Everything points to `P2_00` as being the correct pin, but it doesn’t work for some reason.

dmcleish91, 2021-04-12:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6297973/Marlin.zip)

cr20-123, 2021-05-26:
Not sure if this was ever fixed but when I still used this board I would have to use `#define NO_SD_HOST_DRIVE` in order to get the sd detect to work. I also had to do this on various other boards.

thisiskeithb, 2021-07-15:
> I would have to use #define NO_SD_HOST_DRIVE in order to get the sd detect to work.

Confirmed this fixes the issue. I wonder why mass storage & SD detect on this board conflict.

thisiskeithb, 2021-07-15:
Based on the code block below, Marlin is undefining the `SD_DETECT_PIN` to not interfere with mass storage over USB. This forces you to use the SD mount/unmount option in the LCD.

By enabling `NO_SD_HOST_DRIVE`, you're telling Marlin to not share the SD over USB, so SD detection works.

https://github.com/MarlinFirmware/Marlin/blob/3be35a6bd6845a160c416629ea98392129cd6eb3/Marlin/src/inc/Conditionals_post.h#L424-L438

closed 2021-07-15 by thisiskeithb