MarlinFirmware/Marlin #21603: [BUG] Problems when compiling and connecting the MKS Rovin nano v3 board + MKS-H43

McDAK33, 2021-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Faced several problems:
1. Compilation error;
2. Incorrect display of temperature on the display (see photo);
3. I can not access the USB disk from the computer (writes: insert the disk).

I solved the compilation error by changing the line in the file ...\src\lcd\extui\lib\dgus\mks\DGUSScreenHandler.cpp
> -1274       set_e_relative(true); // M83
> +1274      set_e_relative(); // M83

I doubt this is the right decision.

The problem with connecting the USB disk is solved by sending GCODE: M575 B250. After that I access the disk and it works until the printer is rebooted.

My config files:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6298970/Configuration.zip)

[image]
[image]

### Version of Marlin Firmware

bugfix-2.0.x (2021-04-12)

### Electronics

MKS Robin Nano v3 + TMC2209 + MKS-H43

McDAK33, 2021-04-15:
Hello.
After flashing the current update, the printer completely lost control: the PC connection port does not work (the USB device is not connected correctly), the H43 screen does not start (stops at the boot screen), does not respond to the shutdown button.
If you uncomment the #define MULTI_VOLUME line in the Configuration_adv.h file, and try to compile, you get a lot of errors. After roughly eliminating them (not quite understanding the principle of operation), it is possible to compile the assembly. after the firmware the printer comes to life, but the correct operation of the port can be achieved only by entering the M575 command.

McDAK33, 2021-04-15:
P.S. The temperature on the screen is still not displayed correctly.

thinkyhead, 2021-04-16:
I recently compiled and installed Marlin onto a Nano V3 with an H43 connected, and it all seemed to go well. Unfortunately the H43 I received is damaged so I was unable to test any further.

There have been a number of patches in the last day, some related to temperature, so please check to see if today's bugfix branch has resolved the issue. If it has not, then we can go hunting for some cause.

McDAK33, 2021-04-16:
Have a nice day.
I downloaded the latest update, applied my settings (attached below). The compilation was successful. After flashing, the printer shows no signs of life. Windows reports a device connection error. The H43 screen displays the logo. Can not do anything. I suspect that the new "MULTI_VOLUME" function is to blame for this.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6327280/Config.zip)

McDAK33, 2021-04-19:
Determined the reason for the freeze of my printer. The printer freezes as soon as I connect it to the computer via the USB cable.
Since the printer was constantly connected, I could not figure out the reason. I returned to the bugfix version from 2021-04-12, with this version the connection via USB does not cause problems.

rhapsodyv, 2021-05-02:
The freeze is fixed by #21775 

The serial hang could be related as a bad resend handling from the TFT. We need to look further. I will talk about it with MKS.

McDAK33, 2021-05-03:
I am sorry if I did something wrong. I understand that this is not a primary problem. I hope you remember her.
Thank you.

solawc, 2021-05-05:
I understand this problem. Since the type of temperature display has changed after the update, the screen also needs to be updated synchronously.We made a special version to follow marlin, please go to the link below to download the corresponding FW update
https://github.com/makerbase-mks/MKS-H43/tree/main/DWIN_SET_For_H43 @McDAK33

McDAK33, 2021-05-05:
> Я понимаю эту проблему. Поскольку после обновления тип отображения температуры изменился, экран также необходимо обновлять синхронно.Мы сделали специальную версию, чтобы следить за марлином, перейдите по ссылке ниже, чтобы загрузить соответствующее обновление прошивки
> [https://github.com/ makerbase-mks / MKS-H43 / tree / main / DWIN_SET_For_H43 ](https://github.com/makerbase-mks/MKS-H43/tree/main/DWIN_SET_For_H43) @ McDAK33

I downloaded the firmware for the H43 from the "H43_Firmware_for_Malin" folder. Updated the firmware. Unfortunately, nothing has changed.

https://user-images.githubusercontent.com/80269041/117156947-d8f50c00-adc6-11eb-8cee-44a39d487117.mp4

spilz87, 2021-05-06:
Hello 
I have the same issue with Marlin V2 bugfix and H43 V1.31

After reading codes, it looks like:
- Marlin send a float for actual temperature
- the code of H43 V1.31 is displaying a long 

So I modify the code of H43 to display a float, but still have the issue :(

So I suppose the float format in H43 is different than the one send by Marlin

I don't have the solution, but if that can help to find it ...

closed 2021-07-18 by github-actions[bot]