MarlinFirmware/Marlin #21604: [BUG] PID bed control doesn't use full power

Nach0z, 2021-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Just upgraded from a fairly old version of bugfix-2.0 to whatever the latest is as of 4/12. Reconfigured things as necessary. Bed was previously using bang-bang heating rather than PID control. 
On the previous version, when heating the bed from fairly cold, the power LED on the bed came on, stayed on, and only started blinking when the target temperature was reached.
On the new version, the LED blinks rhythmically, and the bed is heating a fraction as quickly, leading to the bed temp watchdog triggering even with a TEMP_INCREASE of 2 and a TEMP_PERIOD of 300. 
I enabled PID bed control with similar results.
Running M303 E-1 C8 S70 makes the LED light up and stay lit until it starts orbiting the target temperature, as expected, and the bed heats similarly to how it did before upgrading. However, plugging the new Kp, Ki, and Kd values into the bed PID control leads to the same low-power results as stated above.

Downgrading to my previously installed firmware results in the heated bed performing exactly as expected. I have not yet had the time to try and go back through previous versions of marlin to replicate this issue. 

BANG_MAX, PID_MAX, and MAX_BED_POWER are all set to 255. 
When disabling PIDTEMPBED and enabling BED_LIMIT_SWITCHING, the behavior does not visibly change, and the LED continues to blink as if it's being PID controlled. Hysteresis limits are ignored and BED_LIMIT_SWITCHING doesn't appear to have any noticeable effect.

Additional notes: I switched the pins for heater #1 (P2_04) and bed (P2_05) in the pins_BTT_SKR_common.h file because of a hardware issue with the bed heater circuit. Power is delivered to the bed through an external FET board, so there's no real power draw on the board itself.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6300337/Configuration.zip)

### Bug Timeline

First started for me today when upgrading to 2.0.7.2, from an unknown earlier version.

### Expected behavior

Heated bed should receive power nonstop during bang-bang bed heating, until hysteresis limit is reached.

### Actual behavior

Heated bed seems to be getting about 50-60% power which isn't enough to get it to 65C on my printer.

### Steps to Reproduce

Download rev: 1742fb86555a33fdcda1468a00a64690e957685a
Disable PID bed control
Enable bed limit switching
Set bed hysteresis to 2
Attempt to set the bed temperature to any reasonable print temperature (65C in my case)
Observe the bed power LED flicker rather than holding steady

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Apr 12 2021 19:22:21) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Hypercube EVO, custom built

### Electronics

SKR V1.3, TMC2130 with software SPI, pi4

### Add-ons

Board is powered by an external FET, which has its own input/output LED indicators reflecting the same information as above.

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2021-04-13:
depending on age of older version In the config folder should be a readme this should list the version.
also in Version.h.
Actually just attach old version zipped if you would, save mucking about.

ellensp, 2021-04-13:
himm... in Marlin/src/module/temperature.cpp   `temp_bed.soft_pwm_amount = temp_bed.celsius < temp_bed.target ? MAX_BED_POWER >> 1 : 0;`
Why is MAX_BED_POWER divided by two?

Nach0z, 2021-04-13:
@ellensp much to my chagrin I don't even have what you've suggested available to me. I took a backup of the output binary prior to upgrading and then did a git pull, so I don't have the old source available to my knowledge.

Nach0z, 2021-04-13:
@ellensp Just on a hunch, I went through my SKR pins config again. My board had an issue with the bed heater circuit so I used an external FET using the second hotend circuit. The SKR common pin config around line 82 has the following:
`#if HOTENDS == 1
  #ifndef FAN1_PIN
    #define FAN1_PIN                       P2_04
  #endif
#else
  #ifndef HEATER_1_PIN
    #define HEATER_1_PIN                   P2_04
  #endif
#endif`

P2_04 is what my bed is using, and it was being bound as FAN1_PIN in this more recent version of the code. So there was some kind of fan PWM shenanigans that were fighting the bed temperature control circuitry and giving me the issues I was having above. More than likely the firmware was trying to keep the cooling fan off, but the bed on, resulting in the rapid flashing described. The reason that M303 was able to set the bed to full power is because the bed autotuning is set to keep the cooling fan on max power for the duration of the tuning, which also means I probably need to re-tune my bed now that the only thing controlling it is the temperature module. I'm not sure why the default is to make the E1 circuit a fan control circuit but really it's my own fault I didn't triple-check the pins I was changing weren't already assigned.

By changing #define FAN1_PIN P2_04  to #define FAN1_PIN P2_05  (the old broken bed heater pin) I managed to resolve the issue and get my heated bed working as intended using bang-bang with bed limit switching AND with pidtempbed. 

Long story short, the issue was more or less my own fault, and if you have a ticket close action that means "user did something stupid and tried to blame everyone else when it broke", I'm sure it would be applicable here. Not sure why MAX_BED_POWER is being divided by 2 in the temperature module but my FET board doesn't seem to be complaining about it, so I'm not going to poke it much.

descipher, 2021-04-15:
> himm, just looking around the code... in Marlin/src/module/temperature.cpp `temp_bed.soft_pwm_amount = temp_bed.celsius < temp_bed.target ? MAX_BED_POWER >> 1 : 0;`
> Why is MAX_BED_POWER divided by two?

Safety feature ... lol, looks like an issue to me. I will test it with  ? MAX_BED_POWER : 0; to verify this is really what it looks like.

closed 2021-04-15 by thisiskeithb

descipher, 2021-04-15:
> Safety feature ... lol, looks like an issue to me. I will test it with ? MAX_BED_POWER : 0; to verify this is really what it looks like.

Nope, not an issue. It's not what it looks like.