MarlinFirmware/Marlin #21605: [BUG] (Dublicate has problem)

ioktem, 2021-04-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

First of all I opened IDEX(total 2 nozzle  opened X_carriage) 3d printer , Sent this code,   
M503 (M218 T1 620,0,0 )
G28 
T0
G1 F2000 X0
T1
G1  F2000 X250    (my bed 500x500)
M605 S2 X250  R0 
M503 (M218 T1 620,0,0 )(  still same value  also opened epprom setting in configuration )
In my opion , anymore extruders have to be same moving but extruder 2 dont move  (Idex is working perfectly)
If I send T1 invalid extruder and no problem but extruders dont move together 
Or what is the cura setting ? My g-code
[message.txt](https://github.com/MarlinFirmware/Marlin/files/6303241/message.txt)

[configuration.adv.txt](https://github.com/MarlinFirmware/Marlin/files/6303187/configuration.adv.txt)
[ramps.h.txt](https://github.com/MarlinFirmware/Marlin/files/6303188/ramps.h.txt)
[configuration.txt](https://github.com/MarlinFirmware/Marlin/files/6303189/configuration.txt

)

### Version of Marlin Firmware

No lcd screen ı have repetier server via raspberry pi 3b+

### Printer model

custom

### Electronics

ramps1.4

### Your Slicer

Cura

### Host Software

Repetier Host

DrumClock, 2021-04-18:
Ano je tam chyba.
Pokud vyberu nástroj T1 a odešlu G1 E100 motor extruderu se otáčí.
Pokud odešlu G1 X100 E100 vozík X2 se pohybuje ale extruder se neotáčí.
JE to ve všech modech IDEX - při pohybu X1/X2 se neotáčí extrudery !

DrumClock, 2021-04-18:
Priložeý G-code  soubor provede test všech modu IDEX ( FULL_CONTROL, AUTO_PARK, DUPLICATION, MIRRORED) 
U mne je to vše v pořádku. (bugfix-2.0.x"  -"2021-04-18)

[IDEX mode test.zip](https://github.com/MarlinFirmware/Marlin/files/6330919/IDEX.mode.test.zip)

ioktem, 2021-04-20:
Does duplicate mode work? Do the extruders move together?

DrumClock ***@***.***>, 18 Nis 2021 Paz, 13:04 tarihinde şunu
yazdı:

DrumClock, 2021-04-20:
Ano, duplicitní režim funguje. 
Ano, extrudery se pohybují společně.

U mně fungují všechny mody IDEX bez problémů.

ioktem, 2021-04-20:
drive etc. Can you upload somewhere? the marlin you use

DrumClock ***@***.***>, 20 Nis 2021 Sal, 09:18 tarihinde şunu
yazdı:

DrumClock, 2021-04-20:
.... je to poslední vydání bugfix 2.0.x ....
úpravy zde: [konfig a rework .zip](https://github.com/MarlinFirmware/Marlin/files/6341362/konfig.a.rework.zip)

closed 2021-07-01 by github-actions[bot]