MarlinFirmware/Marlin #21606: [BUG] Trigorilla + TMC2209 uart mode crashes into X column

pablopeu, 2021-04-13:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Upgraded my drivers to TMC2209 UART on a stock anycubic kossel delta plus, all axis move OK and home OK to the limit switches. extruder moves in the right direction and PID autotunes OK (tried extruding in the air and it works)

BUT

When I start autocalibration, it starts going down, ask to put the probe, then starts touching, at about the 10th touch it goes straight to the X tower and crashes into it. Made a video of this here: https://www.youtube.com/watch?v=FnrUWFzFqVo

Never managed to finish autocalibration, so, after a reset I auto homed again and moved the Z axis slowly to touch paper and Z on the display shows a value of around 165 which seems wrong

I don't know what parameter to adjust to avoid the autocalibration column crash. Any idea?

Config files are here 
[Marlin delta 2.072.zip](https://github.com/MarlinFirmware/Marlin/files/6304738/Marlin.delta.2.072.zip)

Thanks

### Version of Marlin Firmware

2.072

### Printer model

 anycubic kossel delta plus

### Electronics

trigorilla, tmc2209 uart

### Host Software

OctoPrint

thisiskeithb, 2021-04-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-13 by thisiskeithb

pablopeu, 2021-04-13:
I updated the post with more info as I managed to compile bugfix, is this still considered user support?