MarlinFirmware/Marlin #21608: [FR] (restart message to show current Z height)

MasterCATZ, 2021-04-13:
### Is your feature request related to a problem? Please describe.

I keep getting heater failed  please restart messages , but then I loose what the current print height was to resume printing 

### Are you looking for hardware support?

tronxy

### Describe the feature you want

can the message also show current x,y,z locations so I know where it was last printing so gcode can be modified for a resume ?

another issue is how can I get Z to home in a location that their is no 3d object in the way when using sensor instead of switches , every-time I get x,y homed and relocated to a safe position , trying to home z sends it right in the middle where the print is ?

thinkyhead, 2021-04-15:
> how can I get Z to home in a location

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Tom's 3D Forums](https://forum.toms3d.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

thinkyhead, 2021-04-15:
> but then I [lose] what the current print height was

First: Solve the "heater failed" errors. You can modify the firmware yourself for the feature you're requesting, but I don't think we'll make it a general feature. If "heater failed" errors are happening, it's a bad idea to start more printing before fixing the machine.

I would suggest you enable `POWER_LOSS_RECOVERY` because it is able to continue after a "heater failed" error.

fedetony, 2021-04-29:
Maybe you could use REALTIME_REPORTING_COMMANDS, they will give auto-report of the position while moving. You can also pause and resume and stop the machine. You need to enable also:
EMERGENCY_PASRSER, FULL_REPORT_TO_HOST_FEATURE, HOST_KEEPALIVE_FEATURE, NANODLP_Z_SYNC and NANODLP_ALL_AXIS. :) Have fun

closed 2021-11-28 by thisiskeithb