MarlinFirmware/Marlin #21609: [BUG] (bug summary)

supernielsen, 2021-04-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a SKR PRO v1.2 board.

using the lates BUGFIX marlin i get some errors while i compile.

### Bug Timeline

NEW bug when i like to change to lates bugfix branch to enable usb support

### Expected behavior

I expected it to compile without errors.

### Actual behavior

Marlin\src\HAL\STM32\HAL.h:54:21: error: 'MSerial0' was not declared in this scope; did you mean 'MSerial10'?

Marlin\src\HAL\STM32\../../inc/../core/serial.h:110:74: error: template argument 1 is invalid

Marlin\src\HAL\STM32\../../inc/../core/serial.h:129:47: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}

### Steps to Reproduce

1. make configuration
2. Build configuration with env SKR_PRO_USB_FLASH_DRIVE

https://ibb.co/9V4tCBs
https://ibb.co/5MQdmHX
https://ibb.co/vYtknQp
https://ibb.co/P5KtBWG

### Version of Marlin Firmware

LATEST BUGFIX 020008

supernielsen, 2021-04-13:
[image]
[image]
[image]
[image]

ellensp, 2021-04-14:
Trigger is when you set a serial port to -1 on that env.

ellensp, 2021-04-14:
Looks to be deliberate, perhaps usb_flash_drive and a usb cdc serial are incompatible. 
In env:BIGTREE_SKR_PRO_usb_flash_drive
is `build_unflags     = -DUSBCON -DUSBD_USE_CDC` which disabled usb cdc serial

thisiskeithb, 2021-04-14:
> Looks to be deliberate, perhaps usb_flash_drive and a usb cdc serial are incompatible.

Correct.

You have to move jumpers on the board as well when deciding which feature to enable:
[image]

closed 2021-04-14 by thisiskeithb