MarlinFirmware/Marlin #21610: menu not updating when canceling SD print during dwell (G4)

jmerc77, 2021-04-13:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

when stoping an SD print during dwell (G4), print appears to be stopped until I look at the menu. it still has the pause, stop, etc.

### Bug Timeline

noticed during a 6 minute dwell (G4) of SD print

### Expected behavior

when I select stop print, menu should also go back to the normal "in-idle" options and not act like its still printing. regardless of current G or M code.

### Actual behavior

G4 is preventing the menu from going back to the "in-idle" option set. i.e. looks to still be printing even though the main screen shows it is stopped.

### Steps to Reproduce

add "G4 S360" to end of gcode on SD
print
during the 6 minute dwell navigate to stop and stop the print.
once back to main screen, go back into the menu, and notice the options listed don't match up to expected.

### Version of Marlin Firmware

2.0.7.2

### Printer model

cr10s

### Electronics

skr v1.3, stock display, bigtreetech relay v1.0

### Add-ons

skr v1.3

### Your Slicer

Cura

### Host Software

SD Card (headless)

thisiskeithb, 2021-04-14:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

jmerc77, 2021-04-14:
is there a tool to convert my config to the bug fix branch? I did a lot of tweaking an don't remember it all.

thinkyhead, 2021-04-15:
> is there a tool to convert my config to the bug fix branch? I did a lot of tweaking and don't remember it all.

You'll have to use your favorite text editor and diffing tool to compare your configurations to the original, unmodified configs. I hear Notepad++ has a nice plugin. I'm not aware of any utility that can convert older configurations to newer ones.

jmerc77, 2021-04-15:
just now tried the bugfix and problem is still there. tested with these 2 lines of gcode:
G28 X Y Z
G4 S 360; dwell

jmerc77, 2021-06-15:
I know its a minor bug but, is there going to be a fix? being able to properly cancel the dwell is handy for an auto power off script that can be canceled.