MarlinFirmware/Marlin #21613: [BUG] CUSTOM MENU: return and audible not working (WITH SOLUTION FOUND)

BaBBa71, 2021-04-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I use "#define CUSTOM_MENU_ITEM" in Configuration_adv.h (last 2.0.7.3 bugfix) and enable 

` #define CUSTOM_MENU_MAIN_SCRIPT_AUDIBLE_FEEDBACK // 
  #define CUSTOM_MENU_MAIN_SCRIPT_RETURN           // `

nothing happened (LCD no return to main menu after script execution and not sound played).

SOLUTION:
I have found a mismatch in main_menu.cpp at line 104 where the function:

`#if ENABLED(CUSTOM_MENU_MAIN)

  void _lcd_custom_menu_main_gcode(PGM_P const cmd) {
    queue.inject_P(cmd);
    TERN_(MAIN_MENU_ITEM_SCRIPT_AUDIBLE_FEEDBACK, ui.completion_feedback());
    TERN_(MAIN_MENU_ITEM_SCRIPT_RETURN, ui.return_to_status());
  }`

contains another variable defined (no CUSTOM_etc but MAIN_etc).

@thinkyhead can you fix the menu_main.cpp ?

Thank you in advance.

### Version of Marlin Firmware

2.0.7.3 bugfix

### Printer model

custom delta printer

### Electronics

mega2560

thisiskeithb, 2021-04-14:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

BaBBa71, 2021-04-14:
I've submitted a Pull Request as suggested by @thisiskeithb

closed 2021-04-14 by BaBBa71