MarlinFirmware/Marlin #21614: Thermal Runaway Protection - Safely Park Hotend

marcdraco, 2021-04-14:
### Is your feature request related to a problem? Please describe.

I have a couple of different printers (TronXY and Ender 3 v2) which are repeatedly suffering from thermal runaway protection issues.

Whatever the underlying cause (I think it's the room temp confusing the PID routines because the heater can't keep the hotend hot enough) both machines have stopped fairly early in a print and left a huge "snotball" where leftover plastic is oozing out and smushing into the existing extruded layer.  This doesn't just ruin an otherwise decent print, it's hard to remove and leaving it there cause problems of the hotend crashing into it when the print is resumed.  This affected the Ender 3 so badly that the (albeit loose) drive belt jumped several mm causing a huge and unrecoverable layer shift. 

TL;DR

Would it be possible to "park" (home?) the printer at least a few mms clear of the print and ideally well away from the where the printing is being done - I assume there's safe zone, perhaps where the waste line is printed - so it's possible to correct and re-start a partial print without having to deal with the melted blob?

### Describe the feature you want

Park the hotend away from the print when the PID reports thermal runaway.

Fxguy1, 2021-04-19:
I have also had some thermal runaway faults due to a loose temp sensor in the hot end. One happened at 7 hours into an 8 hour print! Instead of simply shutting down and forcing a reset, could the firmware (once it detects a thermal runaway) automatically pause the print and set the temp target on the hot end to 0? 

If it would simply pause the print, set the target temp to 0 then a simple fix of swapping out the temp sensor or simply re-inserting it into the hot end could be performed and then the print resumed without issue.

marcdraco, 2021-04-19:
In my case it was Marlin beating the **** out of the default firmware (not that I would point *any* fingers at TronXY) (cough). I'd removed the silicone sock and the printer was unable to keep up. Rather worryingly, the printer had been working without a hitch up until that point, even stopping when a wire snapped from the thermistor. The build quality of these things is, er, questionable in places.

But yeah, I think as FXGuy notes that's also a good solution - provided the nozzle is homed or otherwise raised away from the print otherwise we end up with a blob of melted plastic.

marcdraco, 2021-04-19:
I would assume (but correct me if I'm wrong) that the idea is that the hotend is closed down. In fact, parking the hot end *away* from the print is liable to reduce the risk of fire. Recall that the runaway is only targetting the heater(s) in this scenario, so moving the print head away from flammable plastics is better than leaving it to cook.

Runaways only cause fires when they get out of control like the Anet A8s were IIRC - so the idea is to pull the plug on anything generating heat. 

We can't account for fires caused by other parts of the device - a stepper motor, wiring or even (heaven forfend) the PSU itself. Any of these things can (and probably will) cause a fire if allowed to continue unabated which is why we should never leave our printers unattended - although most of us do. 

I suspect we all should really invest in some sort of fire prevention - a smoke detector is a start! (And with that, I'm going to quietly slink into a corner and demand to know why I haven't put one on my rigs yet!)

Fxguy1, 2021-04-20:
I’m not saying disable the safety feature. I’m saying handle it better so that, whatever the cause, the issue can be fixed and then the print resumed. Its reading from a G-Code file, why on earth could it not output a recovery G-Code file starting at the line last executed just like it outputs a log file? 

I’m saying fine, deal with the safety issue, but don’t completely ruin a print that’s been running for days and thus wasting filament and time.

fedetony, 2021-04-20:
If your sensor disconnects, your machine is not safe. Make sure it will not disconnect anymore by attaching the sensor in a correct way or replace it. The eventuality cost you only a spool of filament, and some print time, much better than a house or a life.

marcdraco, 2021-04-20:
TL;DR - scroll to the end.

Full disclosure, some years back, I improved the designs on safety systems that are designed to save lives in a situation where a single spark could ignite a gas mixture; so I'm very well aware of risks. In that case hundreds of people relied on that system working 100% of the time and to date, even the cruder, earlier designs never failed to the point where lives were threatened but there was always a risk that they could.

Fxguy1 and I share the same view, and I'm a little surprised at some of the comments here claiming otherwise.

There's such a thing as a false positive and a good safety system, like the one in the current Marlin, *always* errs on the side of caution. However, I've also seen situations where people find ways to disable the very systems that are there to protect them simply because of this very issue. Disabling this one, for example, only requires a digging around in the code and re-compiling it - not that I imagine *anyone would*, but that's one possible scenario off the top of my head. I assume that it's also possible for the thermistors to fail in such a way that runaway isn't necessarily preventable.

Remember that mission critical monitoring systems always have at least one redundant backup (except the 747-Max but that's a different story).

To wit, the reason I raised this in the first place is for false positives like the ones I faced on not one but two entirely different printer models that we're running some nasty concoction based upon Marlin. I've  read that Creality used to do this but I can't confirm that so it's likely just a rumor.

If we really want to make machines "safe" insomuch as they can be, we need to design in features that the current crop of consumer hardware simply doesn't have. For example gas and thermal detectors capable of shutting down the the mains supply to the printer in the case of fire. Such things exist but they don't come "cheap". Even a smoke detector won't help if you've toddled off to the chemist and left a printer running,

PLA's flash temperature is about 380 C (that's over 700 in old money); for PETG it's in excess of 500 C (>900 F) and ABS comes in at just under 420 C. 
 
It would be interesting to see if it's even possible for a typical cartridge heater to reach that sort of temperature. I assume the most powerful ones intended for more exotic materials will be quite capable of reaching these levels but I'm wondering if anyone actually "did the science". We know that some ANet printers have been blamed for fires - I'd like to see the reports - but ignition requires more than just heat - the fire triangle is heat-fuel-oxygen. Remove one and you don't get a fire.

In the case of an otherwise functioning printer, there's *probably* insufficient oxygen to allow autoignition. However, I could foresee a situation where a glob of spaghetti forms to the extent where that would happen. Wood filament would likely be similarly dangerous but again, we're pushing things to their limits here.

I've heated PLA well beyond its autoignition point and it's not a particularly interesting material to burn. (Since we're having the discussion I'm going to try this a little more later with some dud parts.)

A fire is far more likely to break out elsewhere in the electronics from poorly routed or maintained wiring, so to assume that thermal runaway is going to protect you is like thinking you're as safe in car as you are in the plane.

TL;DR

The reason I started this discussion is not (and never was) with the idea of disabling a protection which is a good thing to have. I'd had false positives from the hotend on both printers because they were lacking a silicone bootie! The low temperature in my gaff meant that the poor thing just wasn't able to keep up without its winter woolies and went on to ruin a bunch of prints at different places, depending on the ambient temp, until I chanced upon the sock issue.

Hence: disable all the heaters (for sure) but park the head and remember the last position so we don't end up with a huge zit and we can restart.

And for heaven's sake, don't rely on software to keep you safe. Use a decent smoke alarm!

marcdraco, 2021-04-20:
Just an update (I'm putting it in a different post for clarity). I figured while designing a simple experiment to see how dangerous PLA would be that the REAL risk for most consumer printers isn't actually fire but the toxic fumes produced by the bowden tube (PTFE) which starts to decompose (pyrolysis) at around 200C but doesn't get "dangerous" until it reaches 260C and the nasty action [fumes that will kill your pet budgie (PTFE toxicosis) and make you feel poorly ] kicks in as it fully breaks down somewhere around the 450C mark at which point your lungs will start to feel pretty rough with "Teflon flu" (https://pubmed.ncbi.nlm.nih.gov/8272977/)

It's not clear if anyone has suffered life altering, long-term effects (or death for that matter) from burning a pan but like anyone with a functional brain, I don't fancy snorting that sort of stuff either. I've certainly been exposed to decomposing PTFE on more than one occasion and I'm still here but that doesn't mean I want to repeat that situation either.

The only death I could find in a quick scan of the literature is detailed here: https://oem.bmj.com/content/73/4/287.1 but you'll note this was an industrial accident, not someone who just left a pan on the stove.

fedetony, 2021-04-20:
My comment was just to say better safe than sorry. Anyway, if you were keen in following the safety rules, you would never leave the machine unattended as specified in most safety instructions.

marcdraco, 2021-04-20:
Absolutely! Sorry if I came over as hostile. It's so hard to get things right in a discussion - things get (cough) heated (cough) very quickly.

But just for ****s and giggles I ran a simple test and pumped an old V6 hotend with a butane torch until the temperature at the block exceeded 400 C measured with a thermocouple where the heater would be. This is well above the temperature that any sane person would be using and beyond most (all?) of the plastics we use I think. I don't do anything fancy because of the costs.

Result: it dripped like a runny nose (and the heat pretty much destroyed the Chinese hotend that came with the TronXY).

Per my suspicion, I was unable to get the stuff to auto-ignite. Now this isn't to say that it cannot happen, only that more tests are needed. I've dropped Stefan (CNC Kitchen) a line to see if he's willing to run a series of tests as I don't have access to a lab or a IR camera - the sort of stuff that would be required to get definitive results.

I have a background in electronics quality control and I suspect those Anet machines caught fire due to poor electrical wiring. In fact, it seems far more likely on the balance of probability and what I know about 3D printers that the true cause of fire is lousy wiring. 

We all know that the experts (the guys who have done this for years) carp on about "wiring management" - and I think there's a lot to be said for that. In fact, I had to fit a strain relief on the Ender 5 *right out of the box* for heaven's sake because the heated bed wire was flopping around all over the place.

The high-current flowing through these wires (and the ones to the hotend) require a lot of small copper wires in a bundle to keep them flexible and capable of handling the power. Over time, if these things get moved around a lot (as they do) there's a very good chance that metal fatigue will set in and the cables will fray. If you saw the Arecibo telescope collapse (https://www.msn.com/en-us/news/watch/arecibo-radio-telescope-collapse-caught-on-video/vp-BB1bBIvM) you'll see how this happens even on huge cables.

Just as those steel cables need most of the threads to retain structural integrity (there's a large margin for error) when enough of them fail, the rest are sure to follow with catastrophic results.

Electrical strain is no different. The cables are rated at specific current and voltage with some amount of headroom to allow for a few strands to fail (I don't design cables but that decision is down to the manufacturer). When enough strands fail the impedance of the wire increases at the junction - in our case, that would most likely be at the connection but it can happen if wires are knotted. As the impedance increases the temperature of the wire increases exponentially, softening the wires that haven't failed and causing them to fail too. So you can see there's a snowball effect here which causes the junction temp to increase to the point where the copper reaches red heat in a matter of seconds (less if the current is sufficient) but in that time, many plastics (as we're discussing) reach auto-ignition temperature and pretty soon you have a fire.

The solution (to some degree) to this is for manufacturers to use self-extinguishing plastics insulation but you can bet the farm that the ****ese probably don't bother doing that. 

Here's an example: https://www.sab-cable.com/products/halogen-free-cables/sabix-sd-745-c-frnc-c1-tp.html - it's probably a bit much for a consumer printer though, it's the sort of stuff you find in commercial aircraft and nuclear power stations according to the makers. 

Probably the sort of stuff you'd find on a Voron though. Damn I'd love to build one of those!

Here's to keeping the hotend just hot enough!

BNNorman, 2021-05-06:
I also get Thermal Protection issues. Sadly I didn't grab an Octopi screen shot showing it but I did note some readings which I think might indicate the PID routine wasn't being called often enough to maintain hot-end temperature during slow printing. (Yes I will bump the speed up again and try again).

The same print errors at the same point in the print run at different times so not a dip in power supply.

I was running a print at 220C. The Octopi serial log shows the Pi was sending "Nxxxx G1 ... " commands but I had lowered the print first layer speed to 10mm/s to improve adhesion on my glass bed  which was steady at 60C steady throughout this.

During the print the hotend temperature started to fall to 207C over a period of 12 seconds. It then started to rise again after 18 seconds. A further 24 seconds later the hotend temperature was 214C then the firmware called a halt. (40 seconds expired).

It looks to me like the PID routine let go of the hotend control.

I could, of course , be totally wrong and I don't know Marlin code well enough to investigate. I just felt that this is worth a mention.

Meanwhile I will increase the THERMAL_PROTECTION_PERIOD slightly to see if the print can ride the dip.

marcdraco, 2021-05-06:
After some more trials on this (and boy it's been busy) I can see why this trigger happens when printing with PLA and a standard E3D V6.

Marlin doesn't know (and I assume that it can't) what sort of hotend we're using so the protection kicks in because we *could* be using a 30, 40 or even 50W heater which is capable (in theory) of driving the hotend to temperatures which could, conceivably and in rare cases cause PLA to catch fire.

I still think this is over-engineering, I guess they can't be too careful.

It's this factor (that the standard V6 with it's stock heater) simply can't get hot enough to autoignite PLA and any danger (be that real or perceived) is more from the gasses emitted from the breakdown from the PTFE lining. "True" all-metal hotends don't suffer with this problem of course, and they can't potentially reach temperatures where even the aluminum block will melt.

thisiskeithb, 2021-12-07:
With https://github.com/MarlinFirmware/Marlin/pull/23172, the hotend will now park on thermal errors.

closed 2021-12-07 by thisiskeithb