MarlinFirmware/Marlin #21615: [BUG] Pause does not work inside an M808 repeat block

kageurufu, 2021-04-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When printing in an M808 repeat segment, the stop and pause functions do not seem to work.

Pausing put Marlin on the "Pausing..." screen, where it then stuck and continued printing multiple loops
Stop was acknowledged, but printing continued regardless

The filament runout sensor was ignored as well.

All of these functions were tested and work fine outside an m808 on the same firmware

Full firmware for the this release is tagged at https://github.com/kageurufu/Marlin-CR30/releases/tag/bugfix-2.0.x_20210413-085553
The test g-code used for this print is at https://static.grid.space/gcode/cr30-level.gcode

### Bug Timeline

Old issue

### Expected behavior

Pausing or Stopping will stop the print

### Actual behavior

Print continues in the loop ignoring pause/stop

### Steps to Reproduce

1. Print a G-Code with an m808 loop segment
2. Pause the print during the looping portion

### Version of Marlin Firmware

bugfix-2.0.x_20210413-085553

### Printer model

Creality CR-30

### Electronics

Stock

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

neilvangeffen, 2021-10-02:
Pause works for me, but it ends the M808 loop.
So pause the print, resume works, but at the end of the current loop, it ends.
Filament runout didnt work, but havent tested if it normally works or not

saloid, 2022-05-24:
As quick dirty fix you can comment out [this line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/MarlinCore.cpp#L347).
It resets all loops after pause (and we need to disable it)

Riconec, 2022-12-04:
So no fix?

FlailAway, 2022-12-16:
>> #define CONFIGURATION_H_VERSION 02010200

Still no fix? I am just using simple GCode and it only runs once. Please consider removing this from the Docs. I just wasted half an hour trying to figure out why it was not working and the Docs showed it as valid from v2.0.8.

`
G92
G28
G00 Z20 F4000
GO0 Y20
M808 L4
G00 X250
G00 Y250
G00 X0
G00 Y20
M808
; drops through to here and stops
`

closed 2026-10-11 by thinkyhead