MarlinFirmware/Marlin #21620: [BUG] Incorrect behavior of the DELTA auto-calibration.

Foxies-CSTL, 2021-04-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I own a Delta printer and have merged with the latest BugFix. The compilation is good but I have Delta calibration problems. The auto-home works well but as soon as I launch an autocalibration, the sequence starts well (autohome), the whole goes down to half height and starts probing but in a vacuum. After the first iteration is finished, it announces that it is a failure.
I have checked my settings with an M503 and everything is correct !!
Any idea on the change in sources that causes this?

### Bug Timeline

new bug because with BUMP 21-04-05, all is correct.

### Expected behavior

A normal iteration.

### Actual behavior

The moves are in middle of the height.

### Steps to Reproduce

After flash + M502 and M500, I run a AutoCalibration Delta.

### Version of Marlin Firmware

BUMP 21-04-15

### Printer model

FLSUN QQS-Pro Delta

thisiskeithb, 2021-04-15:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Foxies-CSTL, 2021-04-15:
Add the M11 S247.
[Bug_G33.log](https://github.com/MarlinFirmware/Marlin/files/6319616/Bug_G33.log)

Foxies-CSTL, 2021-04-20:
@thisiskeithb 
No correction, no solution ?

rhapsodyv, 2021-05-02:
can you test #21781 ?

rhapsodyv, 2021-05-02:
The fix was merged

PR #21781 (2021-05-02): Blocking Move Followup 2 (fix delta failed probe)
### Description

Fix a typo in 3517fbdcbdbb3ba790b0aac55e3ffb03e94f3307

I follows the hint of @lujios in #21774 

Co-author: @lujios 

### Benefits

Fix #21767 
Fix #21620
closed 2021-05-02 by thisiskeithb