MarlinFirmware/Marlin #21621: [FR] Beep on boot

Eliminateur, 2021-04-15:
### Describe the feature you want

Add a config flag BEEP_ON_BOOT that makes marlin beep two to tree times on start if set

Can be helpful as an indicator of an abnormal reboot as well

ellensp, 2021-04-15:
You can implement this already without code changes.
Look at STARTUP_COMMANDS
Add some M300 gcodes to play sounds.

Eliminateur, 2021-04-15:
I was actually looking for that before opening the thread thinking "if there's a bootup gcode i can use M300" but only looked at configuration.h and couldn't find anything, did not think to search advanced!.
Perfect, it's just what i was looking for, will put three M300 codes for a triple beep

closed 2021-04-15 by Eliminateur