MarlinFirmware/Marlin #21624: [BUG] X-axis moves to 0 during print movements

Pyro-Fox, 2021-04-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm sorry for the vague description here, but I don't know how else to describe it. While printing this file, either from SD or GCODE (tested via Pronterface for the sake of this bug), the hotend will just jog over to 0 while printing, then go back. It just keeps doing it over and over until cancelled ([see video](https://www.youtube.com/watch?v=mgPEWsVhgyY&t=15s)). I've tried enabling all the logging I could, but I havent the faintest idea what might cause this.

Mind you, I'm only guessing it's moving to 0 based on the video. Note, my `X_MIN_POS` is `-7`.

From watching the video my best guess on the lines that made this movement errors were:
```
[...]
G1 X120.104 Y170.309 E13.0354
G1 X120.104 Y129.691 E16.0751 <- this one
G1 X120.113 Y129.488 E16.0903
[...]
```
```
[...]
G1 X120.704 Y170.297 E12.7887
G1 X120.704 Y129.703 E15.8265 <- this one
G1 X120.712 Y129.525 E15.8399
[...]
```

Video Link (incase you missed it above):
- [https://www.youtube.com/watch?v=mgPEWsVhgyY&t=15s](https://www.youtube.com/watch?v=mgPEWsVhgyY&t=15s)

Logs and GCODE:
- [Gcode and Logs.zip](https://github.com/MarlinFirmware/Marlin/files/6322536/Gcode.and.Logs.zip)

Configuration files:
- _They are on another PC so I need to submit this bug, walk over there, and I will edit this post to add them in a moment_

EEPROM settings
- Search the serial logs for `M503`. I made sure to echo them before testing each time.

### Bug Timeline

New

### Expected behavior

I expect the nozzle to print a straight line from `X120.104 Y170.309` to `X120.104 Y129.691`

### Actual behavior

The nozzle seems to go from `X120.104 Y170.309` to `X0 Y170.309` to `X120.104 Y129.691`

### Steps to Reproduce

It's simply printing the file that causes it

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Apr 15 2021 20:27:11)

### Printer model

Custom Folger Tech FT-5

### Electronics

SKR 1.3 with TMC2209's

### Your Slicer

Simplify3D

### Host Software

Pronterface

thinkyhead, 2021-04-16:
There was a patch for this today.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Pyro-Fox, 2021-04-16:
Well I don't know what happened between 58cc4b7 (the version I first found this issue) and 464b224, but my printer won't load the build. It's just sitting at the bootscreen ☹. I'm going to try building again to see if it's a fluke, and if that fails start working my way back through commits to see which one works.

Pyro-Fox, 2021-04-16:
Found it. 253ee0c is the breaking change.

thisiskeithb, 2021-04-17:
> Found it. #21623 (253ee0c) is the breaking change. My printer will boot fine with any build using a commit prior to that.

Closing since this was fixed in https://github.com/MarlinFirmware/Marlin/commit/bca4652c5e48415cf478da229a253a2106da3e59

closed 2021-04-17 by thisiskeithb