MarlinFirmware/Marlin #21625: [BUG] Regression in 24a095c: printing doesn't follow path described in gcode

salfter, 2021-04-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I went to update the code running my Hypercube 300 today before printing some storage trays.  Shortly after the job started printing, I noticed that the printer was going way off-course while printing the skirt; it was going all the way to the left edge of the bed and then returning to where it should be.  There were roughly 80 commits between my last update and today; I got a list of commits and bisected it to arrive at the commit that sends my printer off-path: [24a095c](https://github.com/MarlinFirmware/Marlin/commit/24a095c5c14b60bcbffc2807d2c8cc8e9af46e90), which says it does something with the math libraries.

I've attached my gcode, which was sliced from [this STL](https://cdn.thingiverse.com/assets/6d/7a/89/11/00/Bin_76x76.stl) with the latest PrusaSlicer.  The same error occurred with another job before this one, and removing ArcWelder from the processing path didn't make any difference either.  I'm printing files from an SD card in the printer, so it's not an issue with OctoPrint (still have a Raspberry Pi hooked up, but it's switched off).

[Bin_76x76_0.2mm_PLA__1h50m.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/6322616/Bin_76x76_0.2mm_PLA__1h50m.gcode.zip)

I've also attached a patchfile that configures recent commits of bugfix-2.0.x for my printer...this made the bisecting process fairly quick:

### Bug Timeline

today

### Expected behavior

See the video at https://home.alfter.us/s/CRAqDHWcq3mxK5F, in which case the skirt is laid down properly before the rest of the print continues.

### Actual behavior

See the video at https://home.alfter.us/s/9ERTgwmbwqm2CXH, where you can see the printhead wander off course as it comes around the back left corner of the skirt.

### Steps to Reproduce

1. Pick a commit halfway through the list.  Configure, build, and flash.  (Good thing my computer builds Marlin in about 20 seconds.)
2. Start a print; observe whether the head stays on course or not.
3. Bisect the list, pick another commit halfway through the part where the regression lies, and repeat until we know which commit to blame.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

homebuilt Hypercube 300

### Electronics

SKR Pro 1.1 with TMC2130 drivers on all axes, plus an I2C EEPROM, an ESP-01S, and a ReprapDiscount graphics display

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

salfter, 2021-04-16:
...and the config patchfile seems to have been dropped on the floor somewhere.  Let's try again.
[marlin-config-hc300-new.patch.zip](https://github.com/MarlinFirmware/Marlin/files/6322672/marlin-config-hc300-new.patch.zip)

salfter, 2021-04-16:
The behavior looks similar to what's described in https://github.com/MarlinFirmware/Marlin/issues/21624...possibly related?

thinkyhead, 2021-04-16:
Definitely possibly.

One of the smaller, seemingly innocuous changes that I made was to add `-fno-signed-zeros` to the build flags. Before I completely revert the experimental NAN patch, please do a test build with that flag removed from `platformio.ini` and run that same G-code again. Of course, send `M111 S8` (dry-run mode) or add it to your G-code so that it won't actually extrude any plastic.

thinkyhead, 2021-04-16:
Also, if you're feeling inclined, also try removing both `-fno-signed-zeros` and `-ffinite-math-only` from the build flags for a second test. The question I'm looking to answer is whether there is a logical error somewhere in the code (based on the false `NAN` value and the use of `ISNAN` instead of `isnan`) or whether the compiler flags are having an effect in some other area that is more difficult to isolate.

Anyway, that commit was definitely experimental, so ultimately I don't mind reverting it until we have a full accounting of its effects.

salfter, 2021-04-16:
> try removing both -fno-signed-zeros and -ffinite-math-only from the build flags

I'll give that a shot later today (I'm at work, and my Hypercube is at home).

salfter, 2021-04-17:
...and it works properly when compiled without ```-fno-signed-zeros``` and ```-ffinite-math-only```.

closed 2023-07-27 by thinkyhead