MarlinFirmware/Marlin #21629: [BUG] (commands P,R,S interfere between GCODE_MOTION_MODES and REALTIME_REPORTING_COMMANDS)

fedetony, 2021-04-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

code does not compile with REALTIME_REPORTING_COMMANDS. Attached a modified parser.cpp that works. or see the changes in #19330 .
[parser.zip](https://github.com/MarlinFirmware/Marlin/files/6326842/parser.zip)

### Bug Timeline

merge

### Expected behavior

marlin compiles

### Actual behavior

marlin gives error:
```
Marlin\src\gcode\parser.cpp: In static member function 'static void GCodeParser::parse(char*)': Marlin\src\gcode\parser.cpp:235:17: error: duplicate case value 235 | case 'S': case 'P': case 'R': { | ^~~~ Marlin\src\gcode\parser.cpp:223:7: note: previously used here 223 | case 'P' ... 'Q': | ^~~~ Marlin\src\gcode\parser.cpp:235:27: error: duplicate case value 235 | case 'S': case 'P': case 'R': { | ^~~~ Marlin\src\gcode\parser.cpp:220:27: note: previously used here 220 | case 'I' ... 'J': case 'R': | ^~~~
```

### Steps to Reproduce

1) add REALTIME_REPORTING_COMMANDS
2) compile

Here is a solution:
```
@@ -217,10 +217,10 @@ void GCodeParser::parse(char *p) {

     #if ENABLED(GCODE_MOTION_MODES)
       #if ENABLED(ARC_SUPPORT)
-        case 'I' ... 'J': case 'R':
+        case 'I' ... 'J':
           if (motion_mode_codenum != 2 && motion_mode_codenum != 3) return;
       #endif
-      case 'P' ... 'Q':
+      case 'Q':
         if (motion_mode_codenum != 5) return;
       case 'X' ... 'Z': case 'E' ... 'F':
         if (motion_mode_codenum < 0) return;
@@ -231,8 +231,37 @@ void GCodeParser::parse(char *p) {
       break;
     #endif // GCODE_MOTION_MODES

+    case 'R': {
+      #if ENABLED(GCODE_MOTION_MODES)
+        #if ENABLED(ARC_SUPPORT)
+          if (motion_mode_codenum != 2 && motion_mode_codenum != 3) return;
+        #endif
+      #endif
+      #if ENABLED(REALTIME_REPORTING_COMMANDS)
+        codenum = 0;                  // The only valid codenum is 0
+        uint8_t digits = 0;
+        while (*p++ == '0') digits++; // Count up '0' characters
+        command_letter = (digits == 3) ? letter : '?'; // Three '0' digits is a good command
+        return;
+      #endif
+    }
+
+    case 'P': {
+      #if ENABLED(GCODE_MOTION_MODES)
+        if (motion_mode_codenum != 5) return;
+      #endif
+      #if ENABLED(REALTIME_REPORTING_COMMANDS)
+        codenum = 0;                  // The only valid codenum is 0
+        uint8_t digits = 0;
+        while (*p++ == '0') digits++; // Count up '0' characters
+        command_letter = (digits == 3) ? letter : '?'; // Three '0' digits is a good command
+        return;
+      #endif
+
+    }
+
     #if ENABLED(REALTIME_REPORTING_COMMANDS)
-      case 'S': case 'P': case 'R': {
+      case 'S': {
         codenum = 0;                  // The only valid codenum is 0
         uint8_t digits = 0;
         while (*p++ == '0') digits++; // Count up '0' characters
~
~
```

### Version of Marlin Firmware

bugfix.2.0.x 20008

thisiskeithb, 2021-04-16:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

fedetony, 2021-04-16:
> Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

thank you, done :)

thisiskeithb, 2021-04-17:
Your fix was merged. Closing.

closed 2021-04-17 by thisiskeithb