MarlinFirmware/Marlin #21631: [BUG] "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support"  MKS Robin E3 v1.1

daBigR--, 2021-04-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I was trying to compile Marlin bugfix-2.0.x branch for my Creality Ender-3 with a MKS Robin E3 v.1.1 board.  In Configuration.h I set serial port to -1, i.e. `#define SERIAL_PORT -1` as indicated in manufacturer [Wiki](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Marlin_firmware).  I also set  ` #define MOTHERBOARD BOARD_MKS_ROBIN_E3_V1_1` .  Upon compilation I get error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."  I'm able to compile Marlin 2.0.6.1 (downloaded from manufacturers' Github) without errors and also Marlin 2.0.7.2 (from marlinfw.org) using v1.0 of the board without errors.  Marlin 2.0.6.1 from manufacturer works OK in the printer.  Marlin 2.0.7.2 from from marlinfw.org compiles OK but Z axis doesn't work because of board v.1.1 different Z axis driver pin assignments not available in that Marlin release.  I'm trying to use bugfix branch because of the need for v1.1 support but I'm getting the error I mentioned before.

### Bug Timeline

new, not present in 2.0.7.2

### Expected behavior

I expected it to compile or at least not get this board error.  

### Actual behavior

Got compilation error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."

### Steps to Reproduce

Set Configuration.h as described and compile

### Version of Marlin Firmware

bugfix-2.0.x downloaded on 20210416

### Printer model

Creality Ender-3

### Electronics

MKS Robin E3 v1.1

### Add-ons

BLTouch

### Your Slicer

Cura

### Host Software

OctoPrint

thisiskeithb, 2021-04-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-16 by thisiskeithb

thinkyhead, 2021-04-17:
The MKS Robin E3 pins file defines `BOARD_NO_NATIVE_USB` so it seems that the error is probably correct.

daBigR--, 2021-04-17:
I guess you may be right.  I mean this is more a support issue than a bug.
The thing is Marlin firmware considers this board not to have native USB
but the vendor in their Wiki page tells you to set Marlin SERIAL_PORT value
to -1. What am I to do since I don't have enough knowledge to contest these
opposing situations.