MarlinFirmware/Marlin #21632: [BUG] #21623 causes boot failure

lightmaster, 2021-04-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

#21623 Prevents booting past the bootlogo. Not sure yet whether this is only with `HYBRID_THRESHOLD` is off or if it always prevents booting. I can remove this commit alone and it will boot fine again.

## Config files
[Marlin 2021-04-16.zip](https://github.com/MarlinFirmware/Marlin/files/6328470/Marlin.2021-04-16.zip)

### Bug Timeline

Started with #21623 PR

### Expected behavior

Marlin should boot completely

### Actual behavior

Marlin gets stuck on the bootlogo

### Version of Marlin Firmware

bugfix-2.0.x as of 2021-04-16

### Printer model

FLsun Q5

### Electronics

BTT SKR 1.4 Turbo

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

thisiskeithb, 2021-04-17:
As I'm updating printers here with the latest `bugfix-2.0.x`, I can confirm that I'm seeing the same issue.

0815Pusher, 2021-04-17:
Hello, confirmed . 
My BTT SKR 1.4 Turbo also not connecting with the last bugfix-2.0.x

viper93458, 2021-04-17:
Same here with SKR Pro 1.0  :(

thisiskeithb, 2021-04-17:
The breaking change was reverted in bca4652c5e48415cf478da229a253a2106da3e59, so download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) again to grab the fix.

closed 2021-04-17 by thisiskeithb