MarlinFirmware/Marlin #21633: [BUG] (bugfix-2.0.x marlin fail to compile with error, buzzer.h:24:10: fatal error: ../inc/MarlinConfig.h: No such file or directory)

twinsenfan, 2021-04-17:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I am trying to update some config in the marlin and rebuild, ran into an error with a config that previously worked last month. The output looks like, seems like some issue with the path of MarlinConfig.h, but I wasn't able to figure which was it was referring to.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6328696/Configuration.zip)
. 

CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.6
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <LiquidCrystal> 1.0.0
Building in release mode
error:Command 'cmd /C dir E:\' returned non-zero exit status 1.
error:Command 'cmd /C dir I:\' returned non-zero exit status 1.

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\feature\bedlevel\ubl\ubl_motion.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\bedlevel\ubl\G29.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\bedlevel\ubl\M421.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\G33.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\M48.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\M665.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\M666.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\config\M304.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\config\M92.cpp.o
In file included from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/../../feature/../lcd/marlinui.h:27,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/trinamic.h:35,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/indirection.h:44,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper.h:47,
                 from Marlin\src\feature\bedlevel\ubl\ubl_motion.cpp:28:
Marlin\src\feature\bedlevel\ubl\../../../module/stepper/../../feature/../lcd/../libs/buzzer.h:24:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
   24 | #include "../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\feature\bedlevel\ubl\ubl_motion.cpp.o] Error 1 

### Expected behavior

Expect marlin to build just fine

### Actual behavior

Error returned:
In file included from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/../../feature/../lcd/marlinui.h:27,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/trinamic.h:35,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper/indirection.h:44,
                 from Marlin\src\feature\bedlevel\ubl\../../../module/stepper.h:47,
                 from Marlin\src\feature\bedlevel\ubl\ubl_motion.cpp:28:
Marlin\src\feature\bedlevel\ubl\../../../module/stepper/../../feature/../lcd/../libs/buzzer.h:24:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
   24 | #include "../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\feature\bedlevel\ubl\ubl_motion.cpp.o] Error 1

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Jan 14 2021 09:57:06)

### Printer model

Kossel Delta

### Electronics

MKS_SGEN_L

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

twinsenfan, 2021-04-17:
A bit more digging, I used a known good config I compiled a couple month ago and got the same error. The only thing that changed might be the VScode platform update somehow broke this.

ellensp, 2021-04-17:
Cannot reproduce on current bugfix using provided configs, updated as needed.

twinsenfan, 2021-04-17:
I am rebuilding the config with the latest bugfix built and will update the test result

twinsenfan, 2021-04-17:
OK, got it built in the latest bug fix but still don't know why it failed. I guess we can close it.

closed 2021-04-17 by twinsenfan

twinsenfan, 2021-04-17:
Finally figured out what the problem was, might not be related to Marlin, but I think it is good to know. So I had my marlin in a folder with path "F:\3D Printing Repo\Kossel Delta MKS GEN_L Repo\My Kossel MKS GEN_L Firmware\Kossel rebuilt - in progress\Kossel - Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\Marlin", which failed to compile. After I move it to folder with shorter name like F:/temp/Marlin-bugfix-2.0.x, it compiled without error.