MarlinFirmware/Marlin #21636: [BUG] (bug summary)Using M33 Command in Marlin Firmware to read the complete file name, gibberish Chinese Characters are appeared?

cappuccino02, 2021-04-17:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Method:
Connect the BIGTREETECH SKR PRO 1.2 3D printer motherboard to the computer via the USB cable, used the serial debugger to send the G code command, and use the M33 command to get the complete file name. Chinese characters appear gibberish.

### Version of Marlin Firmware

2.0.6.1

thisiskeithb, 2021-04-17:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

cappuccino02, 2021-04-19:
Hello，M33 command to get the full file name, Chinese does not display

抄送: "WEJIR   ENCJI   YA     ***@***.******@***.***&gt;;

thinkyhead, 2021-04-19:
It doesn't do so great at English either:
```
M33 /EXTRUD~1/TWEETY/PLA/TW-FAN~1.GCO
> /Extruders/Tweety/???/tw-fan-adapt-CC.gcode
```

cappuccino02, 2021-04-26:
How to use M20 to get the complete file name? And support the use of full file name to print.

抄送: "WEJIR   ENCJI   YA     ***@***.******@***.***&gt;;

 

 M33 /EXTRUD~1/TWEETY/PLA/TW-FAN~1.GCO &gt; /Extruders/Tweety/???/tw-fan-adapt-CC.gcode

thisiskeithb, 2021-11-25:
> How to use M20 to get the complete file name? And support the use of full file name to print.

Enable `LONG_FILENAME_HOST_SUPPORT` and then issue `M20 L`.

That feature was added this summer, so be sure to download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

cappuccino02, 2021-11-26:
You can get a long file name, but it cannot be used for printing, and Chinese cannot be displayed

抄送: "WEJIR   ENCJI   YA     ***@***.******@***.***&gt;;

ellensp, 2021-11-26:
You have never been able to use long file names for printing. This is how it was designed. This is not a bug

cappuccino02, 2021-11-26:
OK, do you support printing with long file names next

抄送: "WEJIR   ENCJI   YA     ***@***.******@***.***&gt;;

closed 2022-02-06 by github-actions[bot]