MarlinFirmware/Marlin #21641: [BUG] Extruder retracting before print

housam-s, 2021-04-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Printer setup: Ender 3 pro with skr mini e3 and bltouch

Issue: has been there since day 1 basically when a print starts, right before the purge line starts the printer retracts heaaaaaps of filament, nothing prints on the purge line, then when the actual print starts it retracts again. Now to get around this, I’ve been feeding the filament back in and the print continues normally. This was happening before any upgrades too. I have posted my start gcode in the comments.

Troubleshooting: 
- Removed all start gcode and it still happens.
- updated the firmware to the latest marlin firmware
- updated firmware to marlin 2.0 bug fix 
- tried another slicer 
- setup a new custom printer profile
- tried manually editing the gcode
- Current running 2.0 bug fix still no change
-

start gcode 

G92 E0 ; Reset Extruder
G28 ; Home all axes
;G29; ABL
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line ***** This is where the issue occurs****
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line  **** issue occurs again*****

### Bug Timeline

This started happening after upgrading to marlin 2.0

### Expected behavior

Expect the extruder not to retract any filament after purge line.

### Actual behavior

Extruder retracting before print 

### Steps to Reproduce

Print anything 

### Version of Marlin Firmware

2.0 bugfix

### Printer model

Ender 3 pro 

### Electronics

Skr mini e3 v2.0 and bltouch 

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2021-04-18:
Please attach your configuration files.

thisiskeithb, 2021-04-18:
I suspect this has something to do with relative vs. absolute positioning for the extruder.

Try adding [`M83`](https://marlinfw.org/docs/gcode/M083.html) before your `G92 E0` line:

```gcode
M83  ; Set extruder to relative mode
G92 E0 ; Reset Extruder
G28 ; Home all axes
;G29; ABL
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
```

housam-s, 2021-04-18:
Have attached my config files here. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6330521/Marlin.zip)

housam-s, 2021-04-18:
> I suspect this has something to do with relative vs. absolute positioning for the extruder.
>
> Try adding [`M83`](https://marlinfw.org/docs/gcode/M083.html) before your `G92 E0` line:
>
> ```gcode
> M83  ; [...]

I will give this one a try and report back.

housam-s, 2021-04-18:
> I suspect this has something to do with relative vs. absolute positioning for the extruder.
>
> Try adding [`M83`](https://marlinfw.org/docs/gcode/M083.html) before your `G92 E0` line:
>
> ```gcode
> M83  ; [...]

Just tried the  M83 command suggested, the issue was not present anymore, however the extruder is just infinitely extruding and not under proper control from the printer. It just kept turning

Eliminateur, 2021-04-22:
I had this exact thing happen to me with several prints and spent and inordinate amount of time trying to fix it, i think i got it working, the issues is the extruder is in absolute mode, do this to your startup gcode, i've changed some of your G1 to G0 for good practice(it's good practice that non-extrusion movements are kept as G0), the zeroing should be after the bed leveling:

G90 ; absolute values(it sets ALL axes to absolute)
G28 ; Home all axes
;G29; ABL
G92 E0 ; Reset Extruder
G0 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G0 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line ***** This is where the issue occurs**** --> this should extrude 15mm alright
G0 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G92 E0 ; Reset Extruder
G1 X0.4 Y20 Z0.3 F1500.0 E15 ; Draw the second line **** issue occurs again***** -->again extrude 15mm
G92 E0 ; Reset Extruder

i'm assuming you're drawing a 200mm long line along Y twice to wipe, imho you don't need such a long wipe, and it also could be spaced more (my own wipe is just 45mm along the X axis with triple steps of extrusion)

housam-s, 2021-04-23:
> I had this exact thing happen to me with several prints and spent and inordinate amount of time trying to fix it, i think i got it working, the issues is the extruder is in absolute mode, do this to your startup gcode, i've changed some of your G1 to G0 for good practice(it's good practice that non-extrusion movements are kept as G0), the zeroing should be after the bed leveling:
> 
> 
> 
> G90 ; absolute values(it sets ALL axes to absolute)
> 
> G28 ; Home all axes
> 
> ;G29; ABL
> 
> G92 E0 ; Reset Extruder
> 
> G0 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> 
> G0 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
> 
> G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line ***** This is where the issue occurs**** --> this should extrude 15mm alright
> 
> G0 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
> 
> G92 E0 ; Reset Extruder
> 
> G1 X0.4 Y20 Z0.3 F1500.0 E15 ; Draw the second line **** issue occurs again***** -->again extrude 15mm
> 
> G92 E0 ; Reset Extruder
> 
> 
> 
> i'm assuming you're drawing a 200mm long line along Y twice to wipe, imho you don't need such a long wipe, and it also could be spaced more (my own wipe is just 45mm along the X axis with triple steps of extrusion).
> 
> 
> 
> I don't have the exact gcode i used at my printer at hand here, but i'll be able to check it later today

Thank you so much. Will give this a try!

erikcramerjr, 2021-05-17:
> Thank you so much. Will give this a try!

Did this fix your issue I have something similar on my SKR Mini E3 board. but it does not happen every time. 
Only if I change filament right before a print.

Boia11, 2021-07-10:
> Just tried the M83 command suggested, the issue was not present anymore, however the extruder is just infinitely extruding and not under proper control from the printer. It just kept turning

Same problem with my Anet A8 with no firmware change (I use configuration file example). I use version 2.0.9.1.
At starting print the extruder start to infinitely retract filament

danielenestrom, 2021-12-22:
This is happening on my Ender 3 Max with 4.2.2 board and BL Touch. It retracts during purge line instead of extruding. It happens randomly and aborting the print and restarting the same gcode solves the issue most of the times, but as it is seemingly random, sometimes it has this behaviour several times in a row, before behaving correctly. I can see no pattern to it, but I will keep trying to see if I can reproduce the issue with more certainty.

closed 2022-04-12 by github-actions[bot]