MarlinFirmware/Marlin #21644: [BUG] (M203 Max Feedrate not respected when executing feedrate specified in G0-1 command above certain threshold)

charlespick, 2021-04-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When executing G0-1 commands (and possibly others?), Marlin seems to stop limiting feedrate to the feedrate stored in EEPROM or set with m203 at a certain threshold. I suspect this is due to an internal math error or variable overflow. 

### Bug Timeline

Just discovered

### Expected behavior

I expect Marlin to limit the maximum feedrate of an axis to the speed defined by M203 no matter what is specified by F in a G0-1 command to protect the machine from damaging itself. 

### Actual behavior

Marlin attempted to run the axis at the feedrate specified in the G0 command instead of limiting the feedrate to the internal max value, causing the stepper motor to misfunction, make a high pitched noise, and the gantry to barely move. 

### Steps to Reproduce

M203 X5 ; set the feedrate on the x axis to something the steppers should definitely be capable of
G0 X100 F1200 ; move the x axis at 20mm/sec, observe that marlin limits the feedrate to the 5mm/sec as set
G0 X0 F10000000000 ; move the x axis at 166666667mm/sec, observe that marlin does not limit the feedrate properly at speeds above roughly 150000000mm/sec

[video](https://drive.google.com/file/d/1n1qnl3-nB5ZYOaLwWdJkFwgK-A61trjs/view?usp=sharing)
As of posting this, google has not finished transcoding the video, download it from google drive to be able to read what the terminal says, watch with sound on and listen for the stepper drivers high pitched sound

[Serial.log](https://github.com/MarlinFirmware/Marlin/files/6329885/Serial.log)

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6329876/Config.zip)

### Version of Marlin Firmware

2.0.7.2

### Printer model

Creality CR10 V3

### Electronics

Upgraded with BTT SKR Mini E3 V2.0

### Host Software

OctoPrint

ellensp, 2021-04-18:
F10000000000  is 166.667 km/s!   that is faster than the speed of sound!

charlespick, 2021-04-18:
> F10000000000 is 166.667 km/s! that is faster than the speed of sound!

This may be true but the whole point of the feedrate limiter is to protect the machine from insane commands like this getting through. This was discovered when a math bug in a separate piece of software generated this bad command and sent the printer crazy, first completely losing it's location then ramming the gantry into the side of the machine for a few seconds before I could stop it.  Marlin is supposed to prevent this sort of thing from happening just like it prevents the heaters from doing anything that could start a fire.

ellensp, 2021-04-18:
It most definitely needs fixed.

charlespick, 2021-06-20:
Is this fixed?

thisiskeithb, 2021-06-20:
> Is this fixed?

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2021-08-30 by github-actions[bot]