MarlinFirmware/Marlin #21647: [BUG] Library Manager Confliction while building firmware - U8glib-HAL

sarvenn, 2021-04-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While building the firmware a warning about the library manager appears. Which says:

Library Manager: Warning! More than one package has been found by U8glib-HAL @ ~0.4.4 requirements:
 - marlinfirmware/U8glib-HAL @ 0.4.4
 - thinkyhead/U8glib-HAL @ 0.4.4
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts

I don't know if it is a problem or not, just wanted to report it.
@tin

### Bug Timeline

I had built my latest firmware about a month ago, no problem. Latest bugfix2.0 fw has this issue.

### Expected behavior

No confliction messege while building firmware

### Actual behavior

U8glib-HAL confliction

### Version of Marlin Firmware

Latest bugfix2.0 as of today

### Printer model

Ender 3 pro

### Electronics

skr mini e3 v2

### Your Slicer

Prusa Slicer

ellensp, 2021-04-18:
In Marlin/ini/features.ini in bugfix.
simply change  HAS_MARLINUI_U8GLIB     = U8glib-HAL@~0.4.4
to
HAS_MARLINUI_U8GLIB     = marlinfirmware/U8glib-HAL@~0.4.4
if it worries you.

D337z, 2021-04-18:
That directory doesn't exist in bugfix.

thisiskeithb, 2021-04-18:
> That directory doesn't exist in bugfix.

It does:

https://github.com/MarlinFirmware/Marlin/blob/550a806d77201e9a228efd587a92af4adaf648b7/ini/features.ini#L38

thisiskeithb, 2021-04-19:
I just put in a PR to update the PIO package source: https://github.com/MarlinFirmware/Marlin/pull/21659

thisiskeithb, 2021-04-19:
Per https://github.com/MarlinFirmware/Marlin/pull/21659#issuecomment-822152443, closing.

closed 2021-04-19 by thisiskeithb

ellensp, 2021-04-19:
platformio.ini was split up 17 days ago https://github.com/MarlinFirmware/Marlin/pull/21507
Your copy of bugfix must be older than that to not have this file.

sarvenn, 2021-04-19:
Thnx @thisiskeithb