MarlinFirmware/Marlin #21648: [BUG] Serial Communication

fedetony, 2021-04-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With exactly the same configuration from version 20007 the 20008 (new bugfix.2.0.x) the serial communication has issues.
if I have 20007 installed I can flash the machine the first time directly through Visual studio code. After flashing 200008 the computer does not recognize the USB, there is no terminal connection, and I cannot flash it anymore. TFT35 has no connection either, It cannot communicate and shows a no printer attached message. 
I have to remove the SDcard and replace the firmware.bin with a 20007 version, and everything starts working back again normally.
I get the impression that there is something to do with the addition of ethernet in version 20008:
```
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
```
BTW, with the TFT, I've tried to change the baudrate but did not work either. 
Tried also to use it changing serial ports to
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
```
without success. 

### Bug Timeline

new bugfix2.0.x 20008

### Expected behavior

You flash your configured version and you can try to use the machine.

### Actual behavior

It does not connect through serial port.
Computer does not recognize device.
TFT35 can't recognize device is connected.

### Steps to Reproduce

1. Flash version 20008
2. Try to communicate to it.

### Version of Marlin Firmware

bugfix 2.0.x 20008

### Printer model

NA

### Electronics

TFT35, SKR1.4

ellensp, 2021-04-18:
Are you using the absolutely latest bugfix as a few hours ago a bug was fix that stopped skr 1.4 booting (so nothing would run)

fedetony, 2021-04-18:
The issue was yesterday, so if it changed in the last hours, maybe it has changed... I'll check that. Thanks :)

ellensp, 2021-04-18:
I rechecked it, it was yesterday https://github.com/MarlinFirmware/Marlin/commit/bca4652c5e48415cf478da229a253a2106da3e59  So if murphy was watching, this was just after you download marlin

fedetony, 2021-04-18:
master is the bugfix as is, I apply my configurations in a separate branch.
```
FG@FGPC MINGW64 /d/FG Data/SkR1.4 Software/Marlin-bugfix-2.0.x/Marlin-bugfix-2.0.x (master)
$ git checkout FG_Mod_Marlin2.0.x.CNC
Switched to branch 'FG_Mod_Marlin2.0.x.CNC'

FG@FGPC MINGW64 /d/FG Data/SkR1.4 Software/Marlin-bugfix-2.0.x/Marlin-bugfix-2.0.x (FG_Mod_Marlin2.0.x.CNC)
$ git rebase master
Current branch FG_Mod_Marlin2.0.x.CNC is up to date.
```
yet I just downloaded the zip, so I believe my master is not connected to the github repository. How do I connect these two (my local master) and bugfix.2.0.x? 
I just rebased the master, I'll try to flash it soon, my machine is big so I have it somewhere else :)

fedetony, 2021-04-18:
ok, yes, now it worked, apparently I had the older version.

closed 2021-04-18 by fedetony

thisiskeithb, 2021-04-18:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21632