MarlinFirmware/Marlin #21650: [BUG] TMC2130 + Rumba32 TMC Connection Error

Subcode, 2021-04-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,
I'm having trouble using the Rumba32 board and my TMC2130's.
M122 gives the following:
```
Recv: X 0x00:00:00:00 Bad response!
Recv: Y 0x00:00:00:00 Bad response!
Recv: Z 0x00:00:00:00 Bad response!
Recv: E 0x00:00:00:00 Bad response!
Recv: E1 0x00:00:00:00 Bad response!
Recv: E2 0x00:00:00:00 Bad response!
Recv:
Recv:
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: Testing E1 connection... Error: All LOW
Recv: Testing E2 connection... Error: All LOW
```
I have tried this fix: https://github.com/makerbase-mks/MKS-RUMBA32/wiki/About-RUMBA32-TMC-SPI-%22CONNECTION-ERROR%22-FAQ
But i get the same error. No change.
I know for a fact my drivers are set in SPI as they came like that from the manufacturer. 

### Bug Timeline

From install

### Expected behavior

No error during M122

### Version of Marlin Firmware

2.0.7.2 / bugfix-2.0.x

### Printer model

Anet A8

### Electronics

Rumba32

### Add-ons

BL Touch

Subcode, 2021-04-18:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6331322/Marlin.zip)
Config files attached

ellensp, 2021-04-18:
What jumpers do you have set under the stepper driver modules?
what type/brand of stepper drivers (as different brands use different pinouts)

Subcode, 2021-04-18:
Currently i have the pins set like this, this isnt my board but what i used as an example:
https://old.cultivate3d.com/forums/download/file.php?id=225&sid=fc1d74988e6ab61e27e970e850717d90

The stepper driver is a TMC2130 V3.0 from Bigtreetech

Brampie66, 2021-04-24:
Hi,
I am no expert and this may not help you but
On my Aus3D board (V1.1B) TMC/spi just works with all jumpers in the lower position, no wires.
The older boards did seem to have an issue here but at least for the Aus3D board this seems fixed.
I'm using 5160 with steppers from watterott

Subcode, 2021-04-29:
Unfortunately i have a Rumba32 v1.0 from Makerbase, and i tried them all in the lower position without success.

scott-pett, 2021-05-22:
This is a known fault with the V1.0 Aus3D and MKS boards. There is a connection error in the Chip Select track used for the SPI connections.

The workaround is to either modify the board or as I've found, more easily, to mod the TMC2130 driver with a wire link between two pins. 

The Issue detail and fix is described here:  https://github.com/makerbase-mks/MKS-RUMBA32/issues/2

My V1.0 MKS board works fine with TMC2130 with the link in place on the drivers and the TMC2208 drivers work fine in UART mode with the appropriate jumper settings.

The link is easily removed from the driver should you upgrade to a V1.1 RUMBA32 board.

thisiskeithb, 2021-05-22:
Since this is a hardware design issue, I'm going to close this bug report.

closed 2021-05-22 by thisiskeithb

Subcode, 2021-05-22:
As stated above I have already tried this fix...

scott-pett, 2021-05-22:
You do have 12/24volt power applied I presume?

I see TMC connection issue if just connected via USB without the main PSU switched on.

Subcode, 2021-05-23:
I have 12v applied. 
If i do not apply 12v the screen dims. so i know for sure it is powered.
I have tested one of the TMC2130's in a known working board and it started smoking.
Im afraid it is a simple case of fried drivers.