MarlinFirmware/Marlin #21653: [FR] Set Printer Name

KenLucke, 2021-04-18:
### Is your feature request related to a problem? Please describe.

Not really a problem, just making it easier to identify printers when operating a large farm

### Describe the feature you want

M[xx] - Set/Get Printer Name

M[xx] "MyPrinterName" ; would set the name of the printer in UTF8
M[xx] [no parameters] ; would retrieve the previously set name, or "No Printer Name Set" if one had not been set.

charlespick, 2021-05-08:
This is an interesting idea. I almost think it might be a good idea to introduce an optional second name for this feature. I think it's important that there is an unchangeable name compiled into the binary.￼

fugie, 2022-01-16:
This is great idea - and keeping it as simple as possible is also good. @KenLucke has the right recommendation. @thisiskeithb Is anything planned for this? How can this be actioned upon?

thinkyhead, 2025-03-09:
I came across this old FR while particularly bored yesterday. See the starting implementation at the PR linked above. The machine name is reported by various LCD implementations, so those will also need to be updated for the completed feature. The PR provides a good example in a bottle of how to add a G-code to Marlin that has no other UI associated with it.

closed 2025-03-09 by thinkyhead

crossan007, 2025-03-10:
Somewhat tangential to this, but having a "true" UUID would be nice as well without needing to re-compile the firmware