MarlinFirmware/Marlin #21661: Nozzle temperature sometimes cannot reach target - hovers instead about 10 degrees lower until "Heating Failed" timeout

Thinkersbluff, 2021-04-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This bug is present in the CR6-SE Community Firmware at Release 6, which incorporates Marlin bugfix2.0.  I am running this particular version, compiled for my printer's hardware configuration: [CF6.1-Pre2-btt-skr-cr6-with-stock-creality-tft-2021-04-18-22-12.zip](https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6.1-pre2/CF6.1-Pre2-btt-skr-cr6-with-stock-creality-tft-2021-04-18-22-12.zip)

I reported this bug on the Community Firmware GitHub [Issue#248](https://github.com/CR6Community/Marlin/issues/248), but the CF6 developer has asked me to report it upstream, here.

Description:
The nozzle temperature sometimes fails to reach the target value when heating.
When this is happening, the temperature may climb to within a few degrees of target, but then drops again, cycling around a center value approximately 10 degrees below target. Eventually, the system throws a "Heating Failed failed to achieve target temperature within the alloted timeframe" message on the screen and "kills" the job, forcing the user to cycle power to recover.

[image]
[wtf.txt](https://github.com/MarlinFirmware/Marlin/files/6333392/wtf.txt)

### Bug Timeline

Issues with PID not performing as well as in the past have been reported by several users since Release 6 of the Community Firmware.

### Expected behavior

Nozzle should heat to the target temperature and stabilize.  Particularly if the system has just had a PID run (M303 E0 Sxxx U1) at the same target temperature with no problem, yet now can not heat to that xxx temperature to print.

### Actual behavior

When this problem occurs, the serial interface shows that the printer recognizes the correct target temperature, yet it stops short of heating to that value, cycling instead around a value about 10 deg C lower than the target.

NOTE: [These users on the Creality CR6SE/MAX Official Facebook Group describe this same problem](https://www.facebook.com/groups/CR6SECR6MAX/permalink/762752264442045/) in other scenarios, so it is not specifically or uniquely an issue when heating to 235C or when running an esteps extrusion.

I have also been able to successfully achieve 230C when I could not achieve 235C and I have achieved 235C when the nozzle was already at 230C, so there are other parameters at play, here, that I have not yet isolated.  The part cooling fan was off, the whole time.

In the final cycle of the second graph, the printer actually bumped-up from 230 to 235 just at the end, there.  No idea why, as you can see it was settling at the lower value & I touched nothing. 

### Steps to Reproduce

I was able to reproduce this problem fairly consistently as follows:

1. Connect Octoprint to the printer & monitor the serial interface and the Temperature plot to observe what happens
2. Use the CF6 PID function to PID the nozzle at 235C
3. Wait for the nozzle to cool to 145C (or set the temperature to 145C, to let it stabilize there.)
4. Use the CF6 esteps function to run a calibration extrusion at 235C.
5. The printer recognizes and reports on the serial interface that the target temperature is 235
6. The printer stops short of 235 when heating, instead cooling again around 230C
7. The printer then cycles around approx 225 +/-5 degrees, never trying to achieve 235, as seen in the attached logs.
8. If left long enough (sorry, did not time it), the printer throws a Heating Failed alert and kills the process.

### Version of Marlin Firmware

Latest Bugfix2.0 merged into Community FIrmware Release 6.1 Pre 2 on 18 April 2021

### Printer model

Creality CR6-SE

### Electronics

BTT SKR CR6 motherboard, stock hotend, stock cooling fans, stock TFT.  Users with Creality 4.5.3 boards also report this issue.

### Add-ons

None

### Your Slicer

Cura

### Host Software

OctoPrint

Thinkersbluff, 2021-04-19:
I twice edited the above to include this second graph and log, but the template seems to be truncating it out:
[image]
[wtf2.txt](https://github.com/MarlinFirmware/Marlin/files/6333487/wtf2.txt)

Talha909, 2021-04-19:
I have the same issue. I am using 12 April bug fix. Yesterday I try to do PID but its fails. And it does reach to desire temp. It always below 10 degree.

Sebazzz, 2021-04-19:
@Thinkersbluff Have we been able to exclude #21374 being the cause?

portals999, 2021-04-19:
I'm using CR6Comm-CF6-Final-cr6-se-v4.5.3-mb-2021-03-27-15-53 and can confirm I too have this issue, I never saw the problem until I got a new 4.5.3 board and upgraded to CR6.

The temp drops to 10C below what you set it to and hovers around there on the heating and homing screen until the yellow heating error screen appears shortly after.

It is always approx 10C too low and I can consistently reproduce this issue.

scooter2214, 2021-04-21:
Same problem here... Board 4.5.3 and Stock TFT... I use the CR6Comm-CF6-Final-cr6-se-v4.5.3-mb-2021-03-27-15-53.zip  ... 

The temp drops to 10C below what you set it to and hovers around there on the heating and homing screen until the yellow heating error screen appears shortly after.

It is always approx 10C too low and I can consistently reproduce this issue, too 

any solutions ?

Thinkersbluff, 2021-04-21:
> @Thinkersbluff Have we been able to exclude #21374 being the cause?

Nope.  No evidence either way.
 I do suspect, though, that rounding-off the thermistor readings may be causing PID to get “tricked” into cycling around the target temperature minus PID_FUNCTIONAL_RANGE.   That final temperature cycle in wtf2.png above was clearly converging on the lower value then suddenly “flipped” up to the target temp.  I did nothing to cause that. Thermal inertia and random rounding errors could explain that.

Thinkersbluff, 2021-04-22:
Further testing of the CR6 Community Firmware to isolate this issue has determined that:
1. The PID function is not causing the hotend temperature to converge on the target temperature, in version 6.1 Pre2 of the Community Firmware.  Instead, the temperature continues to cycle around an average value that is below target, or aborts the process with a Heating Failed alert. SOMETIMES, the system exits the lower temperature cycling and achieves the target temperature. (In the attached graph, that happens in the first test but does not happen in the third test.  The test in-between shows performing a PID at 235C, when the nozzle was already 235C.)
2. The PID function does work correctly in the previous version (Rel 6 Final) of the same firmware.
3. A GitHub compare of the two firmware releases shows the following list of Marlin 2.0 bugfix commits incorporated at Rel 6 Pre 2: https://github.com/CR6Community/Marlin/compare/a05137d...v2.0.7.3-cr6-community-release-6.1-pre2

Without wishing to exemplify classical "confirmation bias", it does appear as though PR #21374 may have compromised the hotend PID stabilization function.  That change was only merged into the Community Firmware at Release 6 Pre 1, which is where this bug first appeared.

[image]

Thinkersbluff, 2021-04-22:
> @Thinkersbluff Have we been able to exclude #21374 being the cause?
UPDATE - my testing of the Community Firmware Release 6 Prerelease 6 today shows the same behaviour with or without PR #21374 merged into it.  This suggests that this problem pre-dates PR #21374.  

It appears that the problem is related to using the parameters returned by the current PID routine.
[image]
[image]

rhapsodyv, 2021-04-22:
I just sent this to make temperature internal calculations using high precision again, maybe worth a test: #21678

Thinkersbluff, 2021-04-22:
> I just sent this to make temperature internal calculations using high precision again, maybe worth a test: #21678

Thank you.  I just found this existing issue which sounds very similar to the behaviour I am reporting here - https://github.com/MarlinFirmware/Marlin/issues/20463.   

This problem may date back over a year, if related.

thisiskeithb, 2021-04-24:
There have been several temperature-related fixes merged within the last couple days. Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Sebazzz, 2021-04-24:
Sorry, it has become more unstable. Not only "heating failed" errors but thermal runaway as well.

Thinkersbluff, 2021-04-25:
> There have been several temperature-related fixes merged within the last couple days. Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Thank you for the outreach.
@Sebazzz let me know that the latest bugfix2.0 had been merged with the Community Firmware Nightly build of Release 6.1 tonight, so I did download and test it on my system. (I have a Creality CR6-SE with a BTT SKR CR6 v1.0 motherboard and a stock Creality DWIN display.)
My tests did NOT result in any Heating Failed or Heating Runaway alarms today, but the PID tuning performance is still not working well, and the PID Autotune function still seems to make things worse instead of better.

When I was reviewing the closed Issues for possible matches to this problem, I found the description and discussion in [issue 20463](https://github.com/MarlinFirmware/Marlin/issues/20463) was particularly similar and relevant to the situation I think we are having here.  For that reason, I have experimented with two configurations in these tests: 
1. The predefined Configuration.h values of PID_K1=0.95 and PID_FUNCTIONAL_RANGE=10, and 
2. A version incorporating advice from the discussion in that other issue, with PID_K1=0.55 and PID_FUNCTIONAL_RANGE=16

The test procedures that I used and the results that I logged are documented in the .txt file under each plot.

Here is a copy of the configuration files from the second series of tests.  They differ from the first series only in the value of those two parameters.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6371126/Configs.zip)

My limited math skills and lack of C++ literacy is slowing-down my efforts to understand _why_ these changes seem to improve both the hotend stabilization and the PID Autotune function, but the plots seem to confirm that they do.

I would particularly like to understand the significance of editing these parameters, so that I don't recommend changes that only break something else.  I wish I had some way of independently validating the PID factors being generated by Autotune...

[image]
[6.1Nightly24Apr21_PID_K1-95_RANGE-10_DefaultKs.txt](https://github.com/MarlinFirmware/Marlin/files/6371122/6.1Nightly24Apr21_PID_K1-95_RANGE-10_DefaultKs.txt)
[image]
[6.1Nightly24Apr21_PID_K1-55_RANGE-16_DefaultKs.txt](https://github.com/MarlinFirmware/Marlin/files/6371124/6.1Nightly24Apr21_PID_K1-55_RANGE-16_DefaultKs.txt)

pillopaolo, 2021-04-25:
I have a similar problem (temperature 5-10 degrees higher than setpoint) with SKR 1.4 (3 of them), while on SKR 1.3 (4 of them) had never had problems. Same printers (Ender3, Sovol SV01). It seems caused by strange noise/peaks in the thermistor reading.

I improved thing a lot by increasing ADC_LOWPASS_K_VALUE from deafult 2 to 6 or 8.

Reading around and looking at the schematics, I see that SKR 1.4 (and maybe other boards) has different configuration than SKR 1.3, including a ESD Suppressor (CG0603MLC-3.3LE). Might this lead to issues with temperature reading, which need special attention?

Thinkersbluff, 2021-04-25:
In my work life, we found that one of the easiest ways to explore complex technical issues is to exchange highly simplified pictorial models.  If this model is obviously over-simplified in the mind of one or more participant, that suggests an important factor of which other participants may not be aware. By marking up this model to "correct" it, we facilitate a focused conversion of implicit to explicit knowledge.  Through that exchange, we all better understand what matters.  
Currently, I am thinking that PID Autotune is not working on at least some machines, but it seems - because the number of folks reacting to this issue seems low - most Marlin users are not noticing this issue on their own systems. That implies to me that those of us with problems are operating systems that are configured differently somehow from those other users, in a TBD critical way. Figuring out which differences we can ignore and which seem to be relevant is the hard part. 

Open question to those who are interested in solving this problem: Does this model show all of the essential elements that need to be considered + all of the controls available to us to adjust our machine(s) back into a stable operating region?
[image]

Sebazzz, 2021-04-26:
Possibly related #18642 by @bohbotjames

pillopaolo, 2021-04-26:
Adding to my comment above.

If I activate PID debug (#define PID_BED_DEBUG, then M303 D), I see crazy PID output values, especially on the derivative action, when temperature values are noisy and jump +- 1 or 2 degrees is a very short time.

And the fact that the temperature hovers exactly 10 degrees above/below Setpoint is not a coincidence.
Most of the time PID_FUNCTIONAL_RANGE=10!
This means that when above temperature-noise issue is big and PID gives crazy numbers and is not able to control anymore, you end up switching all the time between bang-band and PID modes. The result is a temperature oscillating around [Setpoint - PID_FUNCTIONAL_RANGE] or around [Setpoint - PID_FUNCTIONAL_RANGE].

I did not try, but I bet that if you change PID_FUNCTIONAL_RANGE value, your temperature will hover at a different value.

Thinkersbluff, 2021-04-26:
> I did not try, but I bet that if you change PID_FUNCTIONAL_RANGE value, your temperature will hover at a different value.

This was also my expectation, so I did explore the impact of changing PID_FUNCTIONAL_RANGE. 

The plots posted here show that if I reduce the parameter to 5, the oscillation is more likely to occur consistently.  
If I increase the value to 16, it is more likely to achieve the target temperature.  

Reducing PID_K1 smoothing factor from 0.95 to 0.55 also improved the system’s performance.  Doing both seemed to remove the initial undershoot and ringing almost completely for the conditions tested. 

The ADC_LOWPASS_VALUE parameter is unique to the HAL.h for LPC1768, so not an option for CR6 users on STM32F1 motherboards.

I am only looking at nozzle temperature at the moment but I think the bed and chamber and laser cooler all use the same PID function.

Thinkersbluff, 2021-04-26:
> Possibly related #18642 by @bohbotjames

I agree that looks like the same issue in July 2020. Closed by the author same day without explanation.

Sebazzz, 2021-04-26:
@Thinkersbluff Did we determine if the problem happens on stock Creality boards too? We know the BTT SKR CR-6 board has the temperature noise issue - stock has not.

Met vriendelijke groet,
Sebastiaan Dammann

pillopaolo, 2021-05-01:
Dear developers, it seems that SKR 1.4 (my case, see also **Talha909** ) and CR-6 (see Sebazzz) boards have temperature noise issues, apparently due to their electronics setup.

Why not adding a user configurable filter for the temperature? Something like OVERSAMPLENR, but a bit more powerful and configurable from configuration_adv.h. 

@Thinkersbluff, could you pls reply to above question of Sebazzz? That would be helpful.

Thanks

Thinkersbluff, 2021-05-01:
>@Thinkersbluff, could you pls reply to above question of Sebazzz? That would be helpful.

Which question have I not answered?  These posts seem a little out of sequence but there are at least two stock 4.5.3 MB users above confirming they have this issue.

Thinkersbluff, 2021-05-07:
I updated my Ender 3 with Marlin 2.0.8, yesterday (fantastic work by all, thank you!)
That printer has a BTT SKR E3 Turbo motherboard (a board that is known to have an ADC noise vulnerability in the version I have installed), a MicroSwiss hot end, and a stock cooling system.
I then repeated one of the tests that I had done on my CR6 SE while documenting this PID problem.

In the first graph:
1. I selected Nozzle Temperature of 240C, with the default PID parameters. Notice that there is a short small undershoot and then temperature converges on target and holds +/-1deg.
2. I ran PID by sending M303 S240 E0 C8 U1 via the terminal
3. I repeated step 1, with the new calculated PID parameters. Notice that the system then exhibited stable "ringing" as seen on other systems, but this time it is centered **on** the target value, instead of being centered below the target.

In this test,the PID process itself generated a bit of a messy curve, not the smooth cycling I am used to seeing.  That may somehow correlate to the set of factors that Marlin computed.  The +/- 5-6 degree cycling around the target value with the computed PID parameters persisted for the 10 minutes I left it running.

In the first graph:
1. I cycled power off/on to reset the system and then repeated the first test sequence above, EXCEPT that I did not allow the system to cool as much before I proceeded to the next step.
2. I left that 3rd curve from the first test in this second graph, to highlight the contrast between PID performance with the default PID parameters versus PID performance with the calculated parameters.
3. I noted that the PID Autotune cycling looked much more "normal" in this second run.
4. The calculated PID parameters were quite different in this second run of the test and the nozzle heating was able to converge on target temperature, after a bit of undershooting.

[image]
[image]

Thinkersbluff, 2021-05-07:
Following the advice from team members back in 2016, found on the above related issues, here is the same Ender 3 system as tested in the previous comment, running Marlin 2.0.8 with PID_FUNCTIONAL_RANGE set to 50 (Test conditions and serial log captured in the .txt file below the graph:

[image]
[Ender3_Marlin2_0_8_PID_FUNCTIONAL_RANGE=50.txt](https://github.com/MarlinFirmware/Marlin/files/6443819/Ender3_Marlin2_0_8_PID_FUNCTIONAL_RANGE.50.txt)

I believe this one parameter change could be optimized with experimentation but annecdotal feedback from numerous issue reports seems to support a suggestion that the current default of 10 may be too low for some systems.  This infers to me that Marlin is capable of calculating useable parameters and of performing PID, but it needs a bit more "elbow room" to work, hence the positive impact of increasing the value of that functional range parameter.

esunayg, 2021-05-14:
btt skr cr6 + cf6,1p3
I have been using my device with some little fluctiations, just wanted to make some adjustments I have poked with auto tuning. Acoording to my findings, pid loop doesnt calculate D value correctly. Even if you make d and i to 0 it doesnt overshoot. Normally it should get a fixed k value depending on the target and then over time i should get into the calculation and increase the output value. D comes last to not to overshoot. Any pid device works ok without d value if you dont push k and i way over. Even only K value gives a set value stable response. So you and we are trying to cover the D problem by adjusting the k and i. Thats why auto tune fails to respond. It thinks that D value is changed/iterated but not. Just check with your device while only K value. I and D at 0. Temperature "must" be stabilized at any temp. But it doesnt. Digging deep to understand D value calculation. Either variable doesnt change or miscalculated/miscalled in the Pid pwm calculation loop.

LouwrensMB, 2021-06-07:
I just encountered the same problem my bed temperature head up fine but my nozzle does no reach the set temperature
 was there a solution to this ticket or is it still open
"Heating Failed" timeout #21661

Thinkersbluff, 2021-06-07:
Modifying the PID_FUNCTIONAL_RANGE and PID_K1 factors as described above is at least a proven work-around.

fcaiado, 2021-06-21:
Anyone got a status on a fix for this? Had this issue with an E3v2 and 4.2.2 board and upgraded to the 4.2.7 because I thought it was a board issue then found this thread. This is making printing life horrible and wasting lots of material.

jimberg98, 2021-07-31:
I'm having a similar issue with my bed when PID is enabled. I set the temp and it stops heating 10 degrees shy of the target temp and then says it doesn't reach temp and halts the printer. The LCD indicates that the bed is heating with the wavy line icon, but it clearly isn't. Does it try to use PWM? I have a 1000 watt AC heater which can easily surpass the 100 degrees that it never reaches. I set the temp to 110 for ABS, it makes it to 100 and then halts the printer. I set it to 100 and it halts at 90. I wanted to use PID because I'm having the opposite issue with PID disabled. The temp drifts too high and halts the printer. Someone complained about this a while aga and they just closed it because they figured his bed couldn't reach the temp. I think he was probably having the same issue. I know my bed can reach the temps I'm trying to get to.

jimberg98, 2021-07-31:
Does anyone know why there isn't a separate PID_FUNCTIONAL_RANGE for beds and nozzles? My problem with the bed heatings is clearly related to this. The heater gets shut off when within PID_FUNCTIONAL_RANGE degrees. I would think that a bed which heats much more slowly should have a PID_FUNCTIONAL_RANGE that is much lower. I added a PIDBED_FUNCTIONAL_RANGE to my code and will experiment with it. My bed is a 1/4 inch cast aluminum tooling plate which has a lot of mass to it. Should have mentioned that in my first message.

jimberg98, 2021-08-02:
I got this figured out. PID_FUNCTIONAL_RANGE is the temperature range that PID control takes over. Set your target to 200 and PID control will take over at 190. Up until then, maximum power is supposed to be used... sort-of. The problem occurs because your PID values are set to something that results in a zero power recommendation from a method called get_pid_output_xxx() where xxx is hotend, bed, or chamber. Make sure you have defaults defined and reload your default settings. That will fix the issue with temp never being reached. Or you can just set them with the M codes and save the settings.

Now for the "sort of". For no apparent reason that is documented, the output of get_pid_output_xxx() is divided by 2. Because of this, your heating power will only be running at 50% of max power setting. It would be nice if the code was commented explaining why this is the case. My guess is that since some of the power supplies being put in 3d printers are underrated for the actual power requirements. I can see this being the default behavior for safety, but it should be optional and explained so that people who update their power supplies or stop using them for heating the bed altogether should be able to run at full power. I modified my code so that when it's heating up, it uses full power but then when PID kicks in, I add back the divided by 2 so PID works as designed. My heaters get to temp quickly and the temp control is great. Having a separate PEDBED_FUNCTIONAL_RANGE is also a big help. Using 10 for the hot end and the bed, and probably the chamber, just doesn't make sense. They heat at vastly different rates.

If there is a bug, it's probably that the PID values should be reset to default if using them ever results in a zero power value.

drewskix86, 2021-08-04:
I am having exactly the same issue with my bed, as soon as PID kicks in, the output power drops to nothing and the target temperature is never reached. I've tried PID tuning and using default values but the outcome is always the same. I was using Marlin 2.7.2 without issue previously on the same hardware setup.

edumaker-limited, 2021-08-23:
Having the same issue on Marlin 2.0.9.1, on an Ender-3 v1,  with a Creality 4.2.7 32-bit board.
As described above, bed (bang-bang) heats perfectly to temperature, but the hot end heats to target minus about 8 degrees, then cuts out, drops to target minus 12-15, then reheats but again cuts out around target minus 8.  Tried PID autotune, swapped out heater cartridge and thermistor, tried adding 1000uF capacitor to 5V rail, no joy.  Pleased to see this looks like an issue with the PID implementation, it felt like PID was being interrupted.

Sebazzz, 2021-09-04:
For what it is worth, the PID values determined by the Marlin autotune for my E3D Hemera are:

    M301 P51.5971 I9.3473 D71.2040

These values cause the temperature going up and down endlessly. However, the values below actually work and are stable:

    M301 P34.9800 I3.8300 D79.9200

In both cases `PID_FUNCTIONAL_RANGE` is 25.

avolkov, 2021-10-05:
@thinkyhead The command that saves to EEPROM that you gave fixes the issue -- `M303 E-1 C8 S90 U`

However, I don't think it is the same bug that Creality and BTT SKR 1.4 Turbo users were experiencing in the beginning of the thread.

I was experiencing the same issue with SKR 1.4 Turbo and Marlin 2.0.9.1, and I've tried fixes using ADC_LOWPASS_K_VALUE and PID_KI and PID_FUNCTIONAL_RANGE workaround settings. None of them worked.

I ran M501 and the values for bed Kp Ki Kd were all zero, even I defined them in Configuration.h; here's a sample output --

```
echo:  M301 P20.08 I1.30 D77.33
echo:  M304 P0.00 I0.00 D0.00
```

I think the bug here is that the values should be defaulting from the values defined in Configuration.h and not zeroed out. Also it seems it is currently not possible to load existing BEDPID values, it is only possible to write them using M303.

It could also be that SKR board doesn't honor the code that supposed to load the values.

I had the same issue with:
* Stall sensitivity (M914) for X and Y axis
* Steps per mm (M92)

I suspect this is what users keep filing when they realize they can't use values in Configuration.h when EEPROM is enabled -- https://github.com/MarlinFirmware/Marlin/issues/12468

I'm happy to open a ticket based on this.

It seems all the comments starting on 2021 Jun 21 are referring to EEPROM read issue in 2.0.9.1 rather than PIDTEMP bug with Creality/SKR

ManuelMcLure, 2021-10-05:
So, just to make sure we're on the same page, `M501` will _load_ existing values from EEPROM - it will completely ignore any values set in the firmware configuration values. You need to use `M502` to copy the firmware values into RAM and `M500` to save them back to EEPROM. Only then will `M501` be able to load them back properly.
Apologies if this is something you're already aware of, but I didn't see any mention of `M502` in your comment so I want to make absolutely sure that's not the cause of your issues.

avolkov, 2021-10-05:
@ManuelMcLure Thank you. I'm moving from marlin 1.1.9 when I only sparsely used EEPROM and I  didn't realize I needed to use M502 to load values from firmware.

This seems to be a common misconception, maybe better wording in Configuration.h could alleviate the problem. Referring to Marlin 2.0.9.1, M502 mentions resetting to 'factory defaults' but doesn't mention that defaults also need to be loaded with  M502

ManuelMcLure, 2021-10-05:
Yeah, it's a bit confusing. I always try to explain that there are three levels of configuration in Marlin - RAM, EEPROM, and firmware.
RAM is what the printer will use at runtime. EEPROM is used to initialize the RAM settings on printer boot or if you use `M501`.
Firmware settings will _not_ override EEPROM settings unless:
* You upgrade the firmware to a version that changes the EEPROM _layout_ (i.e. adding or removing EEPROM variables) - if Marlin detects this case it will automatically copy the firmware values to EEPROM and RAM on first boot.
* You explicitly use `M502` to load the firmware values into RAM and then use `M500` to store them, or use the equivalent LCD menus (**Restore Defaults + Store Settings** or **Initialize EEPROM**)

This is done because there's no easy way to detect whether a value was changed in the configuration files if the EEPROM layout didn't change. If you (for example) have updated your Z probe offset and stored it to EEPROM, and then load a new version of Marlin where you forgot to change the Z probe in the configuration, we don't want to override the EEPROM value and possibly cause a nozzle crash.

closed 2021-12-19 by github-actions[bot]

bergie5737, 2022-02-01:
Hi
I've recently upgraded my firmware to latest Marlin bugfix. I  have a Wanhoa I3 clone. In my case my bed was still bang bang when my prints start to fail with "temperature error" as per the display. I replaced the thermocouple on my hotend as I thought that was the issue. I then added PID for the bed and now my bed stop 10C below target. When I do a PID tune, the bed temperature gets to the correct temperature. Any other way of heating stops 10C below target. I am adding this as it seems the issue is not resolved. The hot end is always stable for me.

tombrazier, 2022-03-18:
I might have some capacity to look into this bug. Is there anyone who is actively watching this and could do some testing? @Thinkersbluff  maybe?

In the meantime, if anyone want to experiment with an alternative to PID, I have submitted a PR for model predictive control and I would like to hear back from others how it works for them. #23751

Thinkersbluff, 2022-03-18:
> I might have some capacity to look into this bug. Is there anyone who is actively watching this and could do some testing? @Thinkersbluff maybe?

Yes, I am monitoring this actively and yes, I can make time to run specific tests.  I am an anal-retentive retired Engineer with just enough knowledge and experience to follow instructions rigorously, but I am not a programmer nd I have no access to an electronics lab or to exotic test equipment like oscilloscopes.

I have an Ender3 and a CR6-SE.  
I have modified both printers to direct-drive, with all-metal hotends and 32-bit BTT motherboards. (SKR E3 Turbo on the Ender, SKR CR6 on the CR6).

I do use VSCode/Platformio to compile Marlin 2.x for the Ender, and I can do the same for the CR-6. The display only works if I use the Community Firmware on the CR-6, though, so some controlled experiments may only be possible in “headless mode” on that one. The Ender3 uses the original rotary knob/LCD controller.

I have the stock Type1 thermistors and aftermarket Eigweit 40W heater elements on both printers.  

The CR-6 is currently fitted with a Trianglelabs DragonHF hotend, and I find the aftermarket heater is a marginal fit in that E3 V6 clone heater block. I had to crank the retention screw as tight as I could, to hold the element firmly and  I cannot improve on that, right now. I added thermal paste to improve thermal coupling between heater element and heat block on both printers. The Ender thermistor is a glass bead type, with no thermal paste.  The CR6 thermistor is a cartridge type, with thermal paste and a grub screw.

Both printers do seem to be working, with a small ripple on the extruder temperature but no evidence of this original issue on either machine, “unfortunately”.  One part of any worthwhile experimentation may need to be figuring out how to destabilize the PID control again, before testing a “fix”?

 I do have a PT1000 Type 47 thermistor available for the CR6 but not for the Ender.  The BTT ADC on both printers does seem particularly vulnerable to EMI, and I got frustrated with its inability to stabilize nozzle temperature with that PT1000 thermistor installed so I rolled-back the mod.  
 
How can I be of help?

tombrazier, 2022-03-18:
Hmm. If the error no longer occurs, perhaps there is nothing to do. Are you using a different firmware to the one that generated the graphs above? If so an easy test would be to return to that firmware and see if the problem returns.

Thinkersbluff, 2022-03-18:
> an easy test would be to return to that firmware and see if the problem returns.

I understand.
I have changed both hardware and firmware configurations, since last I reported this behaviour.  Even going back to the previous firmware would not restore my printer to the actual condition it was in, when the above graphs were generated.

I originally posted my graphs and reports on the CR6 Community FIrmware GitHub, when I saw several CR6 Facebook community members chatting about the issue and blaming it on the CF.  We later concluded that the bug was here, upstream of the CF fork, and there certainly do seem to have been a series of bugs over the years, with similar sounding issues.

I can see the temperature readings in the Octoprint Terminal "jumping" up and down by a couple of degrees at a time, from one sample to the next, which "dithering" I imagine is largely a matter of ADC noise overlaid on the "actual" thermal reading digitized with +/- one digit uncertainty at the ADC resolution. I do not understand enough of the system design to know how many significant figures the firmware really has to work with, but there must come a point beyond which it is futile to try to derive more accurate readings from the available data...  

If your PR is designed to improve the ability of Marlin to cope with "noisy" data, is there any value to a log of thermal readings from the terminal on my system? (i.e. do you have a simulation setup to compare the data at the output of various points in Marlin, with and without your PR?)

tombrazier, 2022-03-18:
This is a really fraught area with so many variables. Different sources of error can easily be conflated and what might be a software issue on one machine could look similar to a hardware error on another.

Some of what you describe sounds like #22893. After a long conversation, that issue resulted in two PRs from me which have just been merged, #23871 and #23867. Both could have an effect on the apparent quantization of ADC values. One activates 12 bit ADC (which was supposed to be the norm for 32 bit ARM processors but owing to a subtlety was not) and the other allows 16 times oversampling for when 12 bit ADC is used. On the other hand, maybe the behaviour you mention has a quite different source!

If I had someone who could replicate the 10 degree offset we might be able to work together to find its source.

My MPC PR should do better than PID with noisy data. However I do need real systems to test it against because I have already established that it does well against simulated hotends.

How does the CF work? Does it merge in upstream Marlin changes? Is it similar enough that upstream PRs could be merged easily?

Thinkersbluff, 2022-03-18:
> How does the CF work? Does it merge in upstream Marlin changes? Is it similar enough that upstream PRs could be merged easily?

I believe there is a PR being actively worked now, between @thinkyhead and @Sebazzz to merge the CF fork back into the mainstream. That was the goal of the CF project from the beginning. No idea how long it might take to complete that merge though… Although it may leave support for the stock TFT display on its own branch, most of the CF fork is still Marlin and there is an unreleased extui branch on the CF fork that @Sebazzz has been updating with Marlin PRs.

There may be other CF users who can still reproduce the problem.  I believe at least one of the original Facebook “gang” commented on that issue and may still be monitoring it for updates.  Maybe you can find a useful partner by recruiting on Community Firmware GitHub [issue#248](https://github.com/CR6Community/Marlin/issues/248)

Thinkersbluff, 2022-03-18:
> Some of what you describe sounds like #22893.

Indeed, I believe I also had that problem when I briefly experimented with 2.0.9.2.  I did not have the problem described here, but I could not get Filament Load/Unload to work because the firmware kept waiting for the nozzle temperature to stabilize.  In the end, I swapped the PT1000 out, and the problematic behaviour “disappeared”.