MarlinFirmware/Marlin #21664: [BUG] Extruder 2 movement Z Homing

M4XXX, 2021-04-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a P3Steel with DUAL_X_CARRIAGE (one Head left other right, independent). Reproduce without Slicer.

I use Repetier Server on RPi, here the Log from Terminal to reproduce bug: 
[image]

### Bug Timeline

new

### Expected behavior

Extruder 1 (the second extruder) is selected, homed and somewhere positioned on the buildplate. Now i select Homing Z (only) on Repetier Server. Z Axis is going down and Extruder 2 should remain on same position after homing.

### Actual behavior

Extruder 1 (the second extruder) is selected, homed and somewhere positioned on the buildplate. Now i select Homing Z (only) on Repetier Server. Z Axis is going down. After second touch on Z-Sensor, the Extruder 1 try to go to the Extruder 0 parking Position and collides this it. It goes out of X2_MIN_POS.

### Steps to Reproduce

1. Home X1 and X2 Extruder
2. Select Extruder / X2 and place it somewere on the buildplate
3. Home Z only (after Homing it will drive the X2 in X1)

### Version of Marlin Firmware

2.0.7.2

### Printer model

P3Steel Dual X carriage

### Electronics

Rumba32

### Add-ons

Repetier Server 1.0.4

### Your Slicer

Slic3r

### Host Software

Repetier Host

ellensp, 2021-04-19:
Please attach your bugfix configuration files. So this can be tested and and  check for configuration faults

M4XXX, 2021-04-19:
My configuration files:

[Marlin-2.0.7.2 (2).zip](https://github.com/MarlinFirmware/Marlin/files/6334980/Marlin-2.0.7.2.2.zip)

DrumClock, 2021-04-19:
Ano, je to tak, jen nechápu proč potřebuješ po G28 znovu dělat G28Z ......

M4XXX, 2021-04-19:
I guess its not relevant for pure generated gcode execution, but for me its an issue while calibration. For the second extruder i need to rehome it if i changed the screw for z axis. If X2 / extruder 1 is selected, it will crash into the left x-carriage. 

I dont see the reason why the X axis should move at all while z homing...

DrumClock, 2021-04-19:
@M4XXX  vypadá to, že Marlin předpokládá snímač polohy Z (Bl-Touch) právě na vozíku X1 a proto ho při G28Z také použije.
Bohužel již neví že má zaparkovat vozík X2 a tak se srazí ..........

Já mám kalibraci vozíku X2 nastavenou v parametru :
```
#define HOTEND_OFFSET_Y { 0.0, -1.20 }  
#define HOTEND_OFFSET_Z { 0.0, -0.60 } 
```
Funguje to perfektně: https://youtu.be/qh_7V5xjoyM

closed 2021-06-30 by github-actions[bot]