MarlinFirmware/Marlin #21666: [BUG] M112 emergency stop  ignored during G12 P1 clean nozzle

dwery, 2021-04-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While executing G12 P1 from OctoPrint terminal, G112 is ignored.

### Version of Marlin Firmware

dda2f58588a0a2fca45aa7cc673f9af020163959

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2021-04-20:
Please attach your configuration files.

fedetony, 2021-04-20:
Your bug description says G112, isn't it M112?
Notice if you put the code differently will not work, as `G112` or `m112` do no have any effect but unknown command response.

dwery, 2021-04-20:
@fedetony good catch, let me try and check if I did the mistake while writing here or in the console...

ellensp, 2021-04-26:
Still no config files?  Did a re-test show anything?

dwery, 2021-04-28:
@fedetony you were right, it worked. I can’t recall why I did but most probably I punched in the code manually and mistakenly used the wrong one.

closed 2021-04-28 by dwery