MarlinFirmware/Marlin #21669: [BUG] Resuming from M600 filament change extrudes while unparking nozzle

Sebazzz, 2021-04-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When using M600 and the nozzle is parked, when the print is resumed and the nozzle is being unparked a bunch of extra filament is being extruded.

### Bug Timeline

This started with a5d6f6ac9880090d2412bbea0027885b95c98ea2 #21564

### Expected behavior

When the nozzle is unparked after a filament change, no extrusion should take place.

### Actual behavior

After resuming from a parked position, during a M600 filament change, extrusion takes place during the unparking.

[Uploading Github-1.zip…]()

### Steps to Reproduce

1. Run a random print with a M600 in it *or* have M600 configured as filament runout command.
2. Run the print and wait for the M600 *or* trigger it by filament runout
3. [The nozzle is parked and *beep* *beep* *beep*]
4. Load filament by hand or automatically.
5. Resume the print (in my case from ExtUI, but it doesn't seem to matter)
6. [The nozzle is, while it is being unparked, also extruding a lot of filament]

### Version of Marlin Firmware

Latest from master

### Printer model

Creality CR-6 SE

### Electronics

BTT SKR CR6 board; filament sensor (stock)

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Sebazzz, 2021-04-20:
Opened pull request #21670 with bugfix.

Takiora, 2021-04-21:
Hi. 
I had the same problem and found a workaround. 
I don't know if it is the right way but it work well for me with M600 and M125 (and filament runout).
In the file pause.cpp, in the fuction resume_print, i moved the code used to resume position after the position of the extruder has been set.

[resume_print.txt](https://github.com/MarlinFirmware/Marlin/files/6352568/resume_print.txt)

Sebazzz, 2021-04-24:
Has been fixed and merged via #21670

PR #21670 (2021-04-24): Don't extrude while resuming from parked nozzle position after M600 filament change
### Description

Current E-axis position was not taken into account.

### Benefits

Fixes #21669

### Configurations

ADVANCED_PAUSE_FEATURE with PARK_HEAD_ON_PAUSE

### Related Issues

#21669
commit 8b5e7429 (2021-04-24): Prevent extrusion on M600 un-park (#21670)

Fixes #21669
closed 2021-04-24 by Sebazzz