MarlinFirmware/Marlin #21677: Error compiling for board Anet V1.0.

reynum2, 2021-04-22:
When I activate the option for auto level :
`#define AUTO_BED_LEVELING_UBL`
I have a compilation error :

```
Archiving built core (caching) in: /tmp/arduino_cache_818104/core/core_anet_avr_anet_1621df717313d057c92202babd71649a.a
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Marlin.ino.elf section `.text' will not fit in region `text'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: region `text' overflowed by 41954 bytes
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Anet V1.0.
```
With the option : 
`#define AUTO_BED_LEVELING_BILINEAR`
It works fine.

I try to compile the 1.1.9.1 version of Marlin.
with Arduino: 1.8.13 (Linux), Board: "Anet V1.0"

thisiskeithb, 2021-04-22:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-22 by thisiskeithb

ellensp, 2021-04-23:
Just to help a little, it means you have to many options enabled, the created firmware will not fit in the flash on the controller.
Controllers with less than 128 KB of flash are very close to end of life.