MarlinFirmware/Marlin #21681: Missing the job percentage on display

gchek, 2021-04-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Ender 3 pro - the display is lacking the % of job completion. I see a >> 100% but i think this is realted to speed and not very useful.
I see the bar moving but no %

Thanks 

### Bug Timeline

when I moved to Marlin 2.0

### Version of Marlin Firmware

2.0.7.2

### Printer model

Ender 3 pro

### Electronics

v2.4.2 board

### Your Slicer

Cura

### Host Software

SD Card (headless)

gchek, 2021-04-23:
[image]

ellensp, 2021-04-23:
The indicated area is a feed rate percentage.
if you rotate the dial left and right it will go up and down.

ellensp, 2021-04-23:
You need to enable #define DOGM_SD_PERCENT in Configuration_adv.h

closed 2021-04-23 by ellensp