MarlinFirmware/Marlin #21682: [BUG] scara.cpp is not being compiled while using scara configuration (IS_SCARA is not set)

ekare, 2021-04-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Because IS_SCARA is not being set anywhere scara.cpp is not being used and inverse/fw kinematics does not work.

### Bug Timeline

It exists with 2.0.0

### Expected behavior

Scara inverse/fw kinematics should work. 

### Actual behavior

While using scara configuration Marlin behaves like woking with cartesian coordinate system. 

### Steps to Reproduce

Use MP_SCARA/Configuration* 
Compile

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

SCARA

### Electronics

RAMPS 1.4

ekare, 2021-04-23:
When IS_SCARA is set, I get; 

```
/tmp/arduino_build_980905/sketch/src/gcode/motion/G0_G1.cpp: In static member function 'static void GcodeSuite::G0_G1(bool)':
/tmp/arduino_build_980905/sketch/src/gcode/motion/G0_G1.cpp:99:19: error: 'prepare_fast_move_to_destination' was not declared in this scope
       fast_move ? prepare_fast_move_to_destination() : prepare_line_to_destination();
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/arduino_build_980905/sketch/src/gcode/motion/G0_G1.cpp:99:19: note: suggested alternative: 'prepare_internal_move_to_destination'
       fast_move ? prepare_fast_move_to_destination() : prepare_line_to_destination();
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   prepare_internal_move_to_destination

```

ellensp, 2021-04-23:
I used https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/SCARA/Morgan 

Marlin bugfix compiles as expected.

  #define IS_SCARA 1  is called and set
and correct code is selected

```
Compiling .pio/build/mega2560/src/src/module/planner.cpp.o
Compiling .pio/build/mega2560/src/src/module/scara.cpp.o
```

ellensp, 2021-04-23:
you look to be trying to use arduino ide. 
You need to use platformio under vscode. 
https://marlinfw.org/docs/basics/install_platformio_vscode.html

ekare, 2021-04-23:
I will try Platformio (last time I ave used it was bulky with tons of ugly things. 
Does it really differ? 
I am trying to compile MP_SCARA by the way.

ekare, 2021-04-23:
Same think happened. Copied MPSCARA configurations to a clean bugfix version, opened in platformio.. 
Result: 
If I do not manually set IS_SCARA, scara.cpp is not being touched. 
If I manually set; 
```
Compiling .pio\build\mega2560\src\src\gcode\sd\M1001.cpp.o
Marlin\src\gcode\motion\G0_G1.cpp: In static member function 'static void GcodeSuite::G0_G1(bool)':
Marlin\src\gcode\motion\G0_G1.cpp:100:52: error: 'prepare_fast_move_to_destination' was not declared in this scope
       fast_move ? prepare_fast_move_to_destination() : prepare_line_to_destination();
                                                    ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mega2560\src\src\gcode\motion\G0_G1.cpp.o] Error 1

```

ellensp, 2021-04-24:
I just tried MP_SCARA, although it fails to build with an error. "error: 'BUZZ' was not declared in this scope"  Once you add the missing #include "../libs/buzzer.h" it also compiles without issues.

I cannot reproduce this error

1) where did you download code from, is it real marlin? was it today?
2) where did you download Configs from?

ellensp, 2021-04-24:
Although I cannot replicate this exact error. I did get what is probably a related error. Undefined BUZZ, where the required library was not being included as IS_SCARA was not set.

Try this fix.

In Marlin/src/inc/Conditionals_LCD.h is
```
#if EITHER(MORGAN_SCARA, AXEL_TPARA)
  #define IS_SCARA 1
  #define IS_KINEMATIC 1
```

change it to 
```
#if ANY(MORGAN_SCARA, AXEL_TPARA,MP_SCARA)
  #define IS_SCARA 1
  #define IS_KINEMATIC 1
```

Works for me.

ellensp, 2021-04-24:
The PR has been merged.

closed 2021-04-24 by ellensp