MarlinFirmware/Marlin #21684: MKS E3P serial port invalid

Jln1027, 2021-04-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Configuration.h file:

#define SERIAL_PORT -1
#define SERIAL_PORT_2 3

Error: 
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2456:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
     #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."

### Version of Marlin Firmware

NA

### Printer model

Ender 3 pro

### Electronics

MKS ronbin E3P v2.0 

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

ellensp, 2021-04-23:
As the error says
This board does not support a serial port on -1
the usb port is not directly wired to the processor, it goes though a ch240 chip and is thus serial 0.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-23 by ellensp

Jln1027, 2021-04-23:
I'm sorry, I downloaded the files and without modifying anything, this error appears, apparently the serial ports in the configuration file are not well defined