MarlinFirmware/Marlin #21688: [BUG] With DISTINCT_E_FACTORS enabled, the step per unit value for the second extruder is ignored

p-v-n, 2021-04-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have A10M printer and I'm experiencing the following issue:
When enabling DISTINCT_E_FACTORS in Configuration.h and setting different steps per unit for each extruder (mine are little different), I noticed that the value for the first extruder is used for both, and the value for the second is ignored when extruding. It still shows up correctly in the settings on LCD, but changing the value for the second extruder has no effect.
To make this issue obvious for testing purposes, I halved the value for the first extruder, and doubled it for the second in DEFAULT_AXIS_STEPS_PER_UNIT define. The idea is to have extruder 1 extrude 50mm, and extruder 2 extrude 200mm when trying to extrude 100mm from the menu.
I have the latest bugfix-2.0.x checked out (commit a729cdcfdd558d91d977edf70a43d2008586693c).
Rebuild, upload fw, initialize EEPROM, load defaults. Try to extrude 100mm from extruder 1 - it extrudes 50mm - that's expected (I halved the value on purpose). However, trying to extrude 100mm from extruder 2, also extrudes 50mm, where the expectation is that is would extrude around 200mm (because I doubled the value on purpose).
I believe that the settings are stored correctly in EEPROM, and it's a code bug.
 

### Bug Timeline

Just noticed, but I this is the first time I'm trying to enable DISTINCT_E_FACTORS feaature

### Expected behavior

I expect that the value for steps per unit is used respectfully for each extruder.

### Actual behavior

Steps per unit value for first extruder is used for all of them.

### Steps to Reproduce

1. Unpack attached zip with configuration files. [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6369312/Configuration.zip)
2. Build
3. Upload to machine
4. Preheat hotend to allow extrusion
5. Remove Boden tubes from the head
6. Set up that the end of each filament is flush with each tube
5. Set extruder 1 active
6. Extrude 100mm
8. Observe that about 50mm is extruded - that's expected
9. Set extruder 2 active
10. Extrude 100mm
11. Observe that about 50mm is extruded - NOT expected. Expected about 200mm

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Apr 24 2021 01:56:18) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Vadim's A10M EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Geeetech A10M

### Electronics

All stock

### Add-ons

None

### Host Software

SD Card (headless)

Vertabreak, 2021-04-24:
when using mixing there is 1 virtual extruder controlled by vtool so DISTINCT_E_FACTORS does not work with mixing. 
however if you use the single_nozzle system instead it will grant physical control to both extruders and allow DISTINCT_E_FACTORS.
this is not a bug this is simply how these features work making this bug report user error.

p-v-n, 2021-04-24:
> when using mixing there is 1 virtual extruder controlled by vtool so DISTINCT_E_FACTORS does not work with mixing.
> however if you use the single_nozzle system instead it will grant physical control [...]

Thanks for your response. Are you saying that if I disable mixing extruder feature, things should start working as expected?

Vertabreak, 2021-04-24:
disable mixing and enable single_nozzle instead for miss matched extruders mixing requires identical extruders.

p-v-n, 2021-04-24:
> 
> 
> disable mixing and enable single_nozzle instead for miss matched extruders, mixing requires identical extruders.

Thanks! That worked.
Perhaps, MIXING_EXTRUDER and DISTINCT_E_FACTORS should be made incompatible then in SanityCheck.h?

Vertabreak, 2021-04-24:
Pull Request Filed: https://github.com/MarlinFirmware/Marlin/pull/21689

p-v-n, 2021-04-24:
> 
> 
> Pull Request Filed: #21689

Awesome! Thanks!

Vertabreak, 2021-04-24:
PR Merged i think this issue can now be closed.

closed 2021-04-24 by p-v-n