MarlinFirmware/Marlin #21690: [BUG] Issue with re-homing - consecutive homing - x axis only

Brampie66, 2021-04-24:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Hi,

I recently upgraded my printer to Marlin 2.0 bugfix (version from 2 weeks ago).

Everything works ok, except for one thing, G28/ homing.
-If I home after a reset (or the first time after boot) everything works as expected and the machine homes correctly.
-If I re-home, do G28 again, the X axis moves a couple of mm in the wrong direction and then stops, it does not hit the endstop, everytime I do G28 the carriage move a bit to the right.
There is no error, Marlin seems to think it has homed as far as I can tell.

It's very simple cartesian machine, the x endstop is on the left side.
I asume my config is correct as it home well the first time, I tried enabling debugging for LEVELING but can't see any difference between the first and second homeing.

Anyone has any idea ? I tried a couple of things but the issue remains.
Other than this the printer seems to work well.

config is fairly simple:
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG

#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

// @section machine

// The size of the print bed
#define X_BED_SIZE 245 // was 245
#define Y_BED_SIZE 240 // was 250

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 210

/**
 * Software Endstops
 *
 * - Prevent moves outside the set machine bounds.
 * - Individual axes can be disabled, if desired.
 * - X and Y only apply to Cartesian robots.
 * - Use 'M211' to set software endstops on/off or report current state
 */

// Min software endstops constrain movement within minimum coordinate bounds
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  #define MIN_SOFTWARE_ENDSTOP_Z
#endif

// Max software endstops constrain movement within maximum coordinate bounds
#define MAX_SOFTWARE_ENDSTOPS
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
  #define MAX_SOFTWARE_ENDSTOP_X
  #define MAX_SOFTWARE_ENDSTOP_Y
  #define MAX_SOFTWARE_ENDSTOP_Z
#endif

debug output
G28  X0.00 Y0.00 Z0.00
Machine Type: Cartesian
Probe: NONE
Raise Z (before homing) by 5.00
>>> do_blocking_move_to  X0.00 Y0.00 Z0.00
>  X0.00 Y0.00 Z5.00
<<< do_blocking_move_to  X0.00 Y0.00 Z5.00
>>> homeaxis(X)
Home Fast: -367.50mm
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(X, -367.50, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z5.00
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(X, 5.00, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z5.00
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z5.00
...(X, -10.00, 25.00)
<<< do_homing_move  X0.00 Y0.00 Z5.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y0.00 Z5.00 :
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
<<< G28  X0.00 Y0.00 Z5.00

### Bug Timeline

don't know

### Expected behavior

home correctly after every G28 command, not just the first

### Actual behavior

it moved a bit to the right

### Steps to Reproduce

run G28 2 times

### Version of Marlin Firmware

2.0 bugfix

### Printer model

reprap

### Electronics

rumba32

### Add-ons

tmc5160

thisiskeithb, 2021-04-24:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Brampie66, 2021-04-24:
Hi,
thank you for taking a look at this

I think I've done as you asked, I don't have a probe so left out G29
I don't see much difference in the logs between the first and second G28 

thx
[firstHome.txt](https://github.com/MarlinFirmware/Marlin/files/6370344/firstHome.txt)
[secondHome.txt](https://github.com/MarlinFirmware/Marlin/files/6370345/secondHome.txt)
[HOMING_ISSUE.zip](https://github.com/MarlinFirmware/Marlin/files/6370346/HOMING_ISSUE.zip)

Brampie66, 2021-04-24:
I also tried with Marlin 2.0.7.2 and this makes no difference.
probably a config issue on my end ?

sjasonsmith, 2021-11-25:
I'm unable to reproduce this problem. If this problem were universal I would have expected many users complaining about it, which does not seem to be the case.

This likely was a config or mechanical/electrical issue of some sort. Given the number of months which has passed I am going to close this, presuming that the issue has been resolved. @Brampie66, please re-open if that is not the case, and you can still reproduce an issue with the current bugfix version of Marlin.

closed 2021-11-25 by sjasonsmith