MarlinFirmware/Marlin #21693: [BUG] compile error with AUTO_BED_LEVELING_3POINT

zerog2k, 2021-04-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

when using
`PROBE_MANUALLY` and `AUTO_BED_LEVELING_3POINT`
compilation fails with
```
In file included from Marlin/src/gcode/bedlevel/abl/../../../inc/MarlinConfig.h:55:0,
                 from Marlin/src/gcode/bedlevel/abl/G29.cpp:27:
Marlin/src/gcode/bedlevel/abl/G29.cpp: In static member function 'static void GcodeSuite::G29()':
Marlin/src/gcode/bedlevel/abl/G29.cpp:476:69: error: 'class G29_State' has no member named 'abl_points'; did you mean 'abl_probe_index'?
```

### Bug Timeline

worked more than 1 year ago, last time I compiled against bugfix-2.0.x with these options

### Expected behavior

compile without errors

### Actual behavior

compile fails with error

```
In file included from Marlin/src/gcode/bedlevel/abl/../../../inc/MarlinConfig.h:55:0,
                 from Marlin/src/gcode/bedlevel/abl/G29.cpp:27:
Marlin/src/gcode/bedlevel/abl/G29.cpp: In static member function 'static void GcodeSuite::G29()':
Marlin/src/gcode/bedlevel/abl/G29.cpp:476:69: error: 'class G29_State' has no member named 'abl_points'; did you mean 'abl_probe_index'?
```

### Version of Marlin Firmware

n/a compile error

### Printer model

Wanhao Duplicator i3

zerog2k, 2021-04-24:
compile does not fail when using `PROBE_MANUALLY` w/ `AUTO_BED_LEVELING_LINEAR` instead of `AUTO_BED_LEVELING_3POINT`

seems like `abl_points` is only ever defined if `ABL_USES_GRID` is defined
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L100-L122

and I only see `ABL_USES_GRID` defined:
`#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)`
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L844-L846

zerog2k, 2021-04-24:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6370365/configs.zip)

zerog2k, 2021-04-24:
I don't know enough about leveling internals and intended operation to know exactly what the right fix is.
- should/does AUTO_BED_LEVELING_3POINT use grid?
There are several places which imply that these would be related
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L100-L134
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L137-L140

while this implies they are possibly exclusive of each other (i.e ABL_USES_GRID undefined while AUTO_BED_LEVELING_3POINT defined)
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L530-L567
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L604-L683

```
#if ANY(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
  #define ABL_USES_GRID 1
#endif
```
or whether

zerog2k, 2021-04-24:
FWIW, seems to compile fine with release tag `2.0.7.2`, but not w/ `bugfix-2.0.x`

closed 2021-05-02 by thinkyhead

zerog2k, 2021-05-02:
@thinkyhead 
can you advise why this ticket was closed?

thinkyhead, 2021-05-02:
It is patched.