MarlinFirmware/Marlin #21694: [BUG] Error: Build environment 'melzi_optiboot_optimized' is incompatible with BOARD_MELZI.

zerog2k, 2021-04-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

cannot compile platformio envs `melzi_optiboot` or `melzi_optiboot_optimized`
when MOTHERBOARD = BOARD_MELZI

get error
```
Processing melzi_optiboot_optimized (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'melzi_optiboot_optimized' is incompatible with BOARD_MELZI. Use one of these: sanguino1284p_optimized, sanguino1284p, sanguino644p
```

Compiles with `melzi` or `melzi_optimized` ok

It's been over a year since I last compiled this, but as I have loaded optiboot bootloader on this board, I thought that I used melzi_optiboot variants. (Not sure exactly what the difference between melzi and melzi_optiboot is - not finding any source of truth on this at the moment.)

### Version of Marlin Firmware

bugfix-2.0.x branch

### Printer model

Wanhao Duplicator i3

### Electronics

stock melzi clone board

zerog2k, 2021-04-24:
ok, rolling back to release tag `2.0.7.2` it compiles fine using configs from:
https://github.com/MarlinFirmware/Configurations/tree/2.0.7.2/config/examples/Wanhao/Duplicator%20i3%202.1
with env target `melzi_optiboot`

Rolling forward to `bugfix-2.0.x` HEAD fails with the error (using same configs):
```
Processing melzi_optiboot (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
```

thisiskeithb, 2021-04-24:
> Error: Build environment 'melzi_optiboot_optimized' is incompatible with BOARD_MELZI. Use one of these: sanguino1284p_optimized, sanguino1284p, sanguino644p

This tells you to use `sanguino1284p_optimized`, `sanguino1284p`, or `sanguino644p`.

In your case, use `sanguino1284p_optimized`.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-24 by thisiskeithb