MarlinFirmware/Marlin #21695: Archim2 serial communication not stable in Marlin2.0.x bugfix

billyd60, 2021-04-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a bigtreetech TFT 70 V3 on their fw .27 with a customized Taz5 printer and Archim2 mainboard running Marlin 2.0.x bugfix as of 4/8/21

The screen connects via serial for the touchscreen capability (handled by it's own onboard processor). The touchscreen sends  g or m codes via serial to control the printer based on touch options selected (touchscreen does not have to be configured in Marlin). It also connects to the printer by two 10 wire ribbon cables to SC-1 and SC-2 on the Archim2 board for 12864 emulation. That part works without issue.

The serial connection works but yields the occasional error such as unknown command "M305". (I am using a PT100 and amp but they work great regardless of whether I get this error or not) The other bigger issue is printing from the screen's onboard SD card (or USB port) will work for some random amount of time, before the print is eventually ruined by a random move not called for in the Gcode, (a random move or a random extrusion), or the print simply halts. None of this ever occurs in the 12864 emulation mode (ribbon connection). In that mode the prints complete successfully and are well made without defect. But I would like to get the touchscreen working on the serial side.

Archim2 must have the first serial port defined as SERIAL_PORT -1 for usb emulation. And the screen is connected to J28 (on the archim2 board) for it's touchscreen communication. This is defined as SERIAL_PORT_2 0. The binary will not compile if SERIAL_PORT is undefined for the Archim2.

So I suspect the problem is related to issues others have with multiple serial ports and Octoprint.

Things I have tried: Every serial port setting I can think of. Several different shielded serial cables. Rerouting the serial cable. Using the screen without the ribbon connection. Using different SD cards. Disabled linear advance.  Running the Screen by an external power source. Using or disabling stealthchop. 

### Bug Timeline

It's a new bug to me, this is my first time working with Marlin2

### Expected behavior

The serial port connection should be perfect.

### Actual behavior

The serial port connection is inconsistent and random in behavior. (seemingly)

### Version of Marlin Firmware

Marlin bugfix 2.0.x (downloaded from github 4-8-21)

### Printer model

Heavily Modified Taz 5

### Electronics

Archim2.2b Mainboard.

### Add-ons

Bigtreetech TFT70 V3

### Your Slicer

Simplify3D

billyd60, 2021-04-24:
Here are my config file
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/6370666/config_files.zip)

thisiskeithb, 2021-04-24:
> I have a bigtreetech TFT 70 V3 on their fw .27 with a customized Taz5 printer and Archim2 mainboard running Marlin 2.0.x bugfix as of 4/8/21

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

billyd60, 2021-04-24:
> Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Thank you. Do I need to redo the entire setup line by line or can I just copy over my modified files from the firmware I'm currently using?

I changed the following files
pins_archim2.h
variant.cpp
configuration.h
configuration_adv.h

billyd60, 2021-04-25:
I see a big improvement in serial behavior so far. I will have to run a print and make sure it doesn't die before it finishes. Let you know in a few hours.
Is this just the latest bug fix? If it is I'm shaking my head. If I was here two weeks later with this project I would have saved a lot of gray hair lol. I mean what were the chances that this exact issue got fixed two weeks after I downloaded bugfix on 4/8? It's literally the only problem I was having.

billyd60, 2021-04-25:
There is a noticeable improvement, no more errors. However the print froze after getting through 90% of layer 2.

The printer was non-responsive to commands, but the heaters were still running. I had to cold boot the printer to get rid of the lockup.

I ran it without linear advance disabled, but I am using firmware retract. Stealthchop was disabled for the extruder as well.

billyd60, 2021-04-25:
Tried to run another print via serial but it did have some glitched and then froze again. So while the connection is definitely cleaned up (no more false g or mcode errors at startup) it still has problems.

billyd60, 2021-04-26:
Because Archim2 requires software USB to define SERIAL_PORT_2 0, is this a multiserial problem as defined in [https://github.com/MarlinFirmware/Marlin/issues/21244](url)

billyd60, 2021-04-28:
Ok this has been resolved. The latest version fixed it, the rest of the issue was related to some settings I failed to enable for the BTT TFT 70 V3.0 per their readme on their github here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Sorry for the mistake. I include this info in case someone else makes a similar mistake and perhaps this will save them some time.

closed 2021-04-28 by billyd60