MarlinFirmware/Marlin #21696: Dual z with dual endstops SKR 1.4 homing order

jamesrooke54, 2021-04-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

so here is my issue

I have a corexy, and due to the weight of my bed I need to level my left and right z steppers with an end stop each and when the machine is off the bed falls down unevenly on either side

I have an skr 1.4 and dual z is enabled, on the screen if I issue a move command for z both move 

The problem is that if I issue an auto home command from my click wheel screen then it homes Z and Z2 sequentially and for obvious reasons, I need them to home simultaneously, both endstops are working as i can trigger them manually and they work fine its just the homing order

I don't see any obvious place where I can define the homing order or define that Z and Z2 in marlin itself

I assume i could change my slicer Gcode to set the order but I would prefer if there was a way to define this in the G28 behaviour

### Expected behavior

when auto home is pressed Z and Z2 home simultaneously

### Actual behavior

when auto home is pressed Z and Z2 home sequentially

### Steps to Reproduce

click auto home
Z will home untill endstop is triggered 
then Z2 will begin homing until triggered

### Version of Marlin Firmware

2.0.7.2

### Printer model

BLV CoreXY 

### Electronics

SKR 1.4

### Your Slicer

Cura

### Host Software

Pronterface

jamesrooke54, 2021-05-05:
looks like im going to have to go with a probe instead

ellensp, 2021-05-05:
Please attach your configuration files.

jamesrooke54, 2021-05-05:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/6426595/Configuration_adv.zip)
both config and config adv are here

closed 2021-07-15 by github-actions[bot]