MarlinFirmware/Marlin #21703: PROBING_STEPPERS_OFF causes compile error (in a ender 3 pro / max condig)

moccalotto, 2021-04-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When enabling PROBING_STEPPERS_OFF, I get the following compile errors:

```
Marlin\src\module\probe.cpp: In static member function 'static void Probe::set_probing_paused(bool)':
Marlin\src\module\probe.cpp:249:19: error: 'DELTA' was not declared in this scope
       IF_DISABLED(DELTA, static uint8_t old_known);
                   ^~~~~
Marlin\src\module\probe.cpp:249:26: error: expected primary-expression before 'static'
       IF_DISABLED(DELTA, static uint8_t old_known);
                          ^~~~~~
Marlin\src\module\probe.cpp:249:7: error: 'IF_DISABLED' was not declared in this scope
       IF_DISABLED(DELTA, static uint8_t old_known);
       ^~~~~~~~~~~
Marlin\src\module\probe.cpp:249:7: note: suggested alternative: 'DISABLED'
       IF_DISABLED(DELTA, static uint8_t old_known);
       ^~~~~~~~~~~
       DISABLED
Marlin\src\module\probe.cpp:252:11: error: 'old_known' was not declared in this scope
           old_known = axis_known_position;
           ^~~~~~~~~
In file included from Marlin\src\module\../inc/../core/boards.h:24:0,
                 from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\probe.cpp:27:
Marlin\src\module\probe.cpp:260:20: error: 'old_known' was not declared in this scope
           if (TEST(old_known, X_AXIS)) ENABLE_AXIS_X();
                    ^
Marlin\src\module\../inc/../core/macros.h:89:24: note: in definition of macro 'TEST'
 #define TEST(n,b) (!!((n)&_BV(b)))
                        ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RET6_creality\src\src\module\probe.cpp.o] Error 1
=========================================================================================================== [FAILED] Took 13.62 seconds ===========================================================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:13.619
====================================================================================================== 1 failed, 0 succeeded in 00:00:13.619 ====================================================================================================== 
The terminal process "C:\Users\mocca\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'STM32F103RET6_creality'" terminated with exit code: 1.
```

### Bug Timeline

Unknown. I first noticed the issue today.

### Expected behavior

I would expect the code to either compile or error out in a sensible way. 

### Actual behavior

Compile error

### Steps to Reproduce

use the ender 3 pro 1.5 config.

Enable bltouch and all related settings.

enable PROBING_STEPPERS_OFF

The following link contains BOTH config files even though the title is just Configuration.h

**https://gist.github.com/moccalotto/4495026020881a30a64ad85d23627efc**

### Version of Marlin Firmware

2.0.7.2

### Printer model

Ender 3 Max

### Add-ons

bl-touch

slowbro, 2021-07-03:
I was not able to replicate this, using your configs, nor with the example Ender 3 Max w/ PROBING_STEPPERS_OFF.

thisiskeithb, 2021-07-03:
> I was not able to replicate this, using your configs, nor with the example Ender 3 Max w/ PROBING_STEPPERS_OFF.

It was probably patched a while ago.

closed 2021-07-03 by thisiskeithb