MarlinFirmware/Marlin #21705: [BUG] Any attempts at probing result in "Probing Failed" messages

lightmaster, 2021-04-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On my Delta printer, it's not letting me probe using a Fix Mounted Probe anymore. It's been a while since I've need to use the probe, so no clue when it first started happening, though I haven't changed any settings that should effect probing, so I'm thinking its a bug in a commit.

When I send any command to probe (`M33 P1`, `M48`, `G29 P1`) they all come back with:
```
Error:Probing Failed
Correct delta settings with M665 and M666
ok
```

I with the probe attached, and probe offset temporarily set to 0, I can manually lower the probe to Z0 and that is still the exact value at which the probe activates. Clearly the height setting is correct or else it wouldn't activate exactly at Z0.

Any tips on how to narrow down what's causing it, besides rolling back commits one at a time until it works?

### Bug Timeline

Unknown

### Expected behavior

Probing commands should work successfully when the height is set such that Z0 activates the probe.

### Actual behavior

`Probing Failed` error on all attempts to probe.

### Steps to Reproduce

Not sure, trying to narrow down cause.

### Version of Marlin Firmware

bugfix-2.0.x as of 2021-04-25

### Printer model

FLsun Q5

### Electronics

BTT SKR 1.4 Turbo

### Host Software

OctoPrint

thisiskeithb, 2021-04-27:
> Another statement of problem with the use of fixed probe with the Delta, you have mine with all the logs of operation #21620 

@Foxies-CSTL Someone with the proper hardware to test will get to your issue when they have time, so please hang tight and not confuse issues.

thisiskeithb, 2021-04-27:
@lightmaster If you're using real endstops, did you remove the diag pins on your TMCs?

lightmaster, 2021-04-27:
I believe mine to be a duplicate of #21620 based on the description of that issue. I tried searching for "Probing Failed" since that's the error message I get in OctoPrint's terminal and didn't find any existing issue that seemed like mine, and admittedly #21620 has a bit of a vague title.

I will get the debug logs soon as I can can. Mine does however make failed attempts to probe in midair when it reports probing failed, kinda like it thinks it's dropped to the bed and didn't hit the bed when it expected to. It's also dropping faster than it used to without any change to configs related to probing speed.

If I make a zeroed UBL Mesh and then fine tune the entire mesh with paper, it works and moves as expected when printing, getting a nice first layer. It does still fail to probe even then though.

lightmaster, 2021-04-27:
> @lightmaster If you're using real endstops, did you remove the diag pins on your TMCs?

Yes. Clipped them with cutters and they do not make any connection. It did work fine with this board previously on a previous version of the firmware, I just don't probe very often so no idea when it stopped working.

Foxies-CSTL, 2021-04-27:
@thisiskeithb 
I'm not mixing the problems! all of them are related to the use of a probe to adjust the geometry of a Delta, and without a reference a Delta doesn't work. It isn't a Cartesian printer that only needs limit sensors.

thisiskeithb, 2021-04-27:
> I just don't probe very often so no idea when it stopped working.

Following the steps [outlined in my comment above](https://github.com/MarlinFirmware/Marlin/issues/21705#issuecomment-827472109) should give us more information.

Vertabreak, 2021-04-27:
encountered a similar problem yesterday that was ultimately resolved by setting the following.
basically the problem was a result of the printer not moving back up to leave clearance space for the redeploy.

```
#define Z_CLEARANCE_DEPLOY_PROBE   20 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  20 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     20 // Z Clearance between multiple probes
#define Z_AFTER_PROBING           20 // Z position after probing is done

#define HOMING_BUMP_MM      { 5, 5, 5 }       // (mm) Backoff from endstops after first bump
#define HOMING_BUMP_DIVISOR { 2, 2, 2 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

```
might be worth trying, hope this might be helpful but if not at least i tried.

thisiskeithb, 2021-04-27:
> I'm not mixing the problems! all of them are related to the use of a probe to adjust the geometry of a Delta

Until they are confirmed to be duplicates of each other, let's not mix the two. Again, someone with the proper hardware to test will get to your issue when they have time, so please hang tight.

lightmaster, 2021-04-27:
> 
> 
> > I just don't probe very often so no idea when it stopped working.
> 
> Following the steps [outlined in my comment above](https://github.com/MarlinFirmware/Marlin/issues/21705#issuecomment-827472109) should give us more information.

Working on it now. Waiting on compiling.

lightmaster, 2021-04-27:
[Probing Failed 2021-04-27.txt](https://github.com/MarlinFirmware/Marlin/files/6384138/Probing.Failed.2021-04-27.txt)

Foxies-CSTL, 2021-04-27:
> encountered a similar problem yesterday that was ultimately resolved by setting the following.
> basically the problem was a result of the printer not moving back up to leave clearance space for the [...]

I just tried it with the latest merge and the above settings. Same probe failure. Also, there are problems with binary generation! It doesn't rename them anymore and its output is a "build.firmware"! But this is another problem.

thisiskeithb, 2021-04-27:
> Also, there are problems with binary generation! It doesn't rename them anymore and its output is a "build.firmware"! But this is another problem.

@Foxies-CSTL: Please open a new bug report.

tritri301, 2021-04-29:
Hi, I also happen to run in this exact issue, I have the FLSUN Q5 running the stock config from the bugfix branch. The printer has the exact same output as @lightmaster I can't make it work no matter the config file I'm using

thisiskeithb, 2021-05-03:
This should be fixed after https://github.com/MarlinFirmware/Marlin/pull/21781

closed 2021-05-03 by thisiskeithb

lightmaster, 2021-05-06:
@thisiskeithb  Will verify that this fixes my issue tomorrow. Crazy that a single line of code can cause so much headache, lol.

lightmaster, 2021-05-07:
I can now successfully problem, thanks.