MarlinFirmware/Marlin #21709: [BUG] Printhead going out of boundaries randomly

doucettom, 2021-04-26:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

First print after bootup or a long pause when it prints, first trial is alway out of boundaries (like the steppers are doubling distance) so it crashes in X-axis belt-tensionner
Cancelling and reprinting the same .gcode file and works fine.
Printing from USB via Octoprint (0.5.3)

I first thought it was related to me slicing with Cura Legacy inside of Octoprint but recently move to Cura 4.8 and it still does that on at least two printers (same motherboard+stepper drivers (2209 UART), MKS SGEN_L V1 but different mecanics)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6377395/Marlin.zip)

### Bug Timeline

2.0.6 up to 2.0.7.2 for sure

### Expected behavior

I expected it to move to print start point (middle of the bed minus print object 1/2 more or less)

### Actual behavior

It looks it moves twice as far (and seemingly twice as fast as well) and crashes into X tensionner

### Steps to Reproduce

1-Boot printer
2-Upload sliced gcode to Octoprint
3-Start printing. Z axis raise a lot higher than it is supposed to and look like all movement are doubled.
4-Cancelling the print
5-Reprinting the same file works fine.

### Version of Marlin Firmware

2.0.7.2 but 2.0.6 did the same

### Printer model

Tronxy X3 and AdimLAB Gantry Pro

### Electronics

both with MKS SGEN_L v1 boards and MKS 2209 UART Steppers

### Your Slicer

Cura

### Host Software

OctoPrint

thisiskeithb, 2021-04-26:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

doucettom, 2021-04-26:
> Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

I will but are you aware of something similar being reported and patched ? I tried to search for something similar to no avail. Most of what I found was ABL-related and I'm not using ABL.

thisiskeithb, 2021-04-26:
> I will but are you aware of something similar being reported and patched

Marlin is updated daily and `2.0.7.2` was released six months ago, so that’s why we ask you to try the latest `bugfix-2.0.x`.

doucettom, 2021-04-26:
Allright. I flashed bugfix release linked above and so far so good. I did only one fake print on the X3 and 2 reals prints on the ADIMLab but still, they went w/o issues.

thisiskeithb, 2021-04-26:
> Allright. I flashed bugfix release linked above and so far so good. I did only one fake print on the X3 and 2 reals prints on the ADIMLab but still, they went w/o issues.

Awesome! I'll close this, but feel free to reply or open a new bug report if you run into any issues.

closed 2021-04-26 by thisiskeithb

doucettom, 2021-04-26:
> > Allright. I flashed bugfix release linked above and so far so good. I did only one fake print on the X3 and 2 reals prints on the ADIMLab but still, they went w/o issues.
> 
> Awesome! I'll close this, but feel free to reply or open a new bug report if you run into any issues.

I'll flash all my MKS_SGEN_L printers to that release and see how it goes in the next few weeks !

doucettom, 2021-04-30:
2.0.8 just released. Should it work ? I've seen some SGEN_L and some homing position stuff in the release notes but the bugfix I tried was pretty recent and I don't know when 2.0.8 was freezed ?

doucettom, 2021-05-11:
It happened again with the bugfix :/ . I can try 2.0.8 if necessary.