MarlinFirmware/Marlin #21712: [BUG] compile error with TEMP_SENSOR_1_AS_REDUNDANT

JohnnyTheOne, 2021-04-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

```
Marlin\src\module\temperature.cpp:424:24: error: conflicting declaration 'int16_t Temperature::redundant_temperature_raw'
   int16_t Temperature::redundant_temperature_raw = 0;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\module\temperature.cpp:34:0:
Marlin\src\module\temperature.h:427:23: note: previous declaration as 'uint16_t Temperature::redundant_temperature_raw'
       static uint16_t redundant_temperature_raw;
                       ^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\temperature.cpp:424:24: warning: declaration of 'uint16_t Temperature::redundant_temperature_raw' outside of class is not definition [-fpermissive]
   int16_t Temperature::redundant_temperature_raw = 0;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\temperature.cpp:425:32: error: conflicting declaration 'celsius_float_t Temperature::redundant_temperature'
   celsius_float_t Temperature::redundant_temperature = 0.0;
                                ^~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\module\temperature.cpp:34:0:
Marlin\src\module\temperature.h:428:24: note: previous declaration as 'celsius_t Temperature::redundant_temperature'
       static celsius_t redundant_temperature;
                        ^~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\temperature.cpp:425:32: warning: declaration of 'celsius_t Temperature::redundant_temperature' outside of class is not definition [-fpermissive]
   celsius_float_t Temperature::redundant_temperature = 0.0;
```

### Version of Marlin Firmware

 bugfix-2.0.x

ellensp, 2021-04-27:
Please test the above PR

JohnnyTheOne, 2021-04-27:
It works thanks

commit 5f9aac20 (2021-04-27): Use temp_info_t for temp_redundant (#21715)

Fixes #21712
closed 2021-04-27 by ellensp