MarlinFirmware/Marlin #21719: [BUG] Not working SKR 2

calderon67, 2021-04-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With latest update SKR 2 board does not even flash itself properly. It never turns on permanent fan (not CNC), TFT always keep on saying that there is no printer connected and so on. Flashing Marlin-pr-btt_skr_v2 from SKR 2 marlin #21503 development does not show any problem (I mean all files that were uploaded on a github page by @thisiskeithb , not only configuration.h and configuration.adv). I am using Visual Studio Code with PlatformIO  and Auto Build Marlin. What is going on?

However, from previous bug reporting, it wasn't working well with TMC2130. Several axes stop working in different moments. At the very beginning it started with the X axes and autolevel feature. When autolevel was commanded without maintaining any temperature on heated bed everything seems to work normally. However, when a bed temperature was commanded, after reaching being maintaining it, suddenly X axes stopped during autolevel.

I also tested to print something (a 10mm cube) without using bed temperature, but suddenly Z axis stop working just like X.

I have commanded M122 for debugging and this was the results:

[image]

I have tried everything. I have tried changing stepper driver, wire, and also stepper motor, but X axes failure during autolevel with heated bed at 70ºC keeps on.

I am pretty sure that it is not a hardware problem. The main reason is that, after getting X failure and when printer thinks that autolevel has been finished, it does not respond to any movement command ordered via Pc or via TFT35. After that, without touching or moving anything, mainly the main board, I press reset button on TFT and after that X axes moves normally once again.

My printer was previously running with ramps with this TMC on SPI, and I used 350ma for X and Y, 700ma for the driver of the two Z stepper motors and 800ma for E. It worked normally with that current set up. My printer is a Prusa i3 with a heated bed of 200mm * 200mm.

Anyway, I have added a 50mm fan just in front of the drivers, and the X driver is the first one on air blow line. Also it should be taken into account that overheat make no sense because it doesn't trigger overheat prewarn tag of TMC2130 and because autolevel is mostly the first moving command, which is not very intense. We are talking about autolevel grid of 9 points.

I do not think it is related with that, but I have not tested to increase ma of the driver. However, why does it suddenly change msteps from 16 to 256? And why does it suddenly change X driver enabled to false?

I have no idea what else to do. I have also contacted BigTreeTech support to confirm that there is no hardware problem, but the fact that everything recovers after reset seems to me that it is a software related problem.

Thank you all for your help!

### Version of Marlin Firmware

Marlin bugfix-2.0.x - 020008

### Printer model

Prusa i3 steel

### Electronics

SKR 2

### Add-ons

TMC2130, TFT35, Bltocuh

### Host Software

SD Card (headless)

### Additional information & file uploads

My configuration files:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6386438/config.zip)

thisiskeithb, 2021-04-27:
> Flashing Marlin-pr-btt_skr_v2 from SKR 2 marlin #21503 development does not show any problem

Start over with a fresh copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and do not mix in or merge in any files from that PR as it changed quite a bit before it was finalized.

calderon67, 2021-04-27:
> Start over with a fresh copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and do not mix in or merge in any files from that PR [...]

I have already used a fresh copy just downloaded from this repository. For configuration I used comparation with Visual Studio tool for changing the downloaded configuration.h and configuration.adv from the fresh copy.

thisiskeithb, 2021-04-27:
Let’s try this:

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

calderon67, 2021-04-27:
> Let’s try this:
>
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

As I have said at the beginning, that marlin version is not been flashed properly on my board. After loading on SD the generated firmware.bin, the board does not even turn on permanent running fan (not CNC but the one next to X stepper motor wires). However, as I have said, if I flash the previous repository from your developing and testing post, it get flashed without any problem. 

So maybe there is something wrong with latest update. I directly downloaded marlin from the link of your post with debug instructions.

What should I do?

Thanks a lot

thisiskeithb, 2021-04-27:
> As I have said at the beginning, that marlin version is not been flashed properly on my board.

The SKR V2 has been confirmed to work with the latest bugfix by several users at this point, so I'm confident you have a hardware failure.

Until then, for best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-27 by thisiskeithb

calderon67, 2021-04-27:
> The SKR V2 has been confirmed to work with the latest bugfix by several users at this point, so I'm confident you have a hardware failure.
>
> Until then, for best results getting help with [...]

But if I have a hardware failure why does it work with the repository that you posted? I mean, why can that be flashed and new one not?

thisiskeithb, 2021-04-27:
> But if I have a hardware failure why does it work with the repository that you posted? I mean, why can that be flashed and new one not?

There is no difference between that merged/now deleted PR & current bugfix. The code from the PR is now part of the main Marlin codebase.

If one copy works and the other doesn't, then there is a mix of bad files somewhere and cannot be a Marlin bug.

calderon67, 2021-04-27:
How can I clean bad files on Visual Studio?

calderon67, 2021-04-30:
> There is no difference between that merged/now deleted PR & current bugfix. The code from the PR is now part of the main Marlin codebase.
>
> If one copy works and the other doesn't, then there is a [...]

Well, after making change version of ststm32 from 12.0 to 12.1 it gets flashed right. Please, open once again the bug, and I will post as soon as possible the requested log file.

ellensp, 2021-04-30:
have you seen https://github.com/MarlinFirmware/Marlin/pull/21741/files

calderon67, 2021-04-30:
> have you seen https://github.com/MarlinFirmware/Marlin/pull/21741/files

No, I don't, what should I do with that? I followed user action mentioned on #21732 and it works now.

calderon67, 2021-04-30:
> Let’s try this:
>
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

I have done what you asked me. I have done a first test without using heated bed without any problem, and after that, I have done another one with heated bed at 70ºC. 

Before sending bed temp command I sent M122 S0 for checking drivers parameters. In this second test it fails, and X stop moving in the middle of autolevel. I sent FAIL to mark the point after which X is not moving although  it is supposed to move. 

At the end of the test I sent M122 S0 once again, and the drivers parameters are not the same. Note that X and Y appear with 256 msteps instead of 16 without any command. They also appear as disabled. I didn't notice that Y axes stopped but it could have stopped at the very end when no other move is expected from Y.

Please let me know any idea or any other test to perform.

Thanks a lot for your help!

[x fail.txt](https://github.com/MarlinFirmware/Marlin/files/6405577/x.fail.txt)

thisiskeithb, 2021-04-30:
Contact BigTreeTech for a replacement: https://twitter.com/BigTreeTech/status/1388080304942985216

calderon67, 2021-04-30:
> Contact BigTreeTech for a replacement: https://twitter.com/BigTreeTech/status/1388080304942985216

How do you know it is not software once again? I remember you that the test have been made with latest bugfix changing version of ststm32 from 12.0 to 12.1 as in #21732

thisiskeithb, 2021-04-30:
> How do you know it is not software once again?

There are known issues with the first revision of this board (as noted in the link above), so please contact BigTreeTech for a replacement.

calderon67, 2021-04-30:
> > How do you know it is not software once again?
> 
> There are known issues with the first revision of this board (as noted in the link above), so please contact BigTreeTech for a replacement.

You are right, I have an SKR 2 rev A, thanks a lot for letting me know the issue.

I would have never known without your help.